trezor/trezor-firmware #3091: Passphrase design

Hannsek, 2023-06-18:
https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?type=design&node-id=8825-68715&t=4gMBHfDOelVRGYl9-4

matejcik, 2023-06-19:
[image]
can't be done.
the decision where the passphrase is entered comes from the host.
(or you can preconfigure it by "force on device")

matejcik, 2023-06-19:
Only one side can decide, the other has to wait for decision.

the conversation goes like this:
Host -> Trezor: `GetAddress`
Trezor -> Host: `PassphraseRequest` (and shows "please enter passphrase on host")
(Trezor waits for response)
Host -> Trezor: `PassphraseAck("helloworld")` **or** `PassphraseAck(on_device=true)`
Trezor -> Host: `Address("tb1foobarbaz")`

an alternative is what we had originally:
Host -> Trezor: `GetAddress`
Trezor -> Host: `ButtonRequest` ( = wait for me to ask the user)
Host -> Trezor: `ButtonAck` ( = i'm waiting for your response)
(Host waits)
Trezor -> Host: `PassphraseRequest` (user decided to enter on host)
Host -> Trezor: `PassphraseAck("helloworld")`
Trezor -> Host: `Address("tb1foobarbaz")`

problem is, the conversation **must** be request/response. Trezor can't up and interject "actually I changed my mind, here's a different response". Technically we could do something about it but it would be a significant change to the whole protocol and Connect would hate us.

(we can consider it for the encrypted session protocol, while we're breaking things anyway. but even then it does make things more difficult for both sides)

mayyyyybe the host could ping the Trezor repeatedly, like, instead of sending the passphrase or "on_device", every 200 ms it sends `PassphraseAck(did the user click the button?)` and the Trezor can either respond `PassphraseRequest` (no they didn't) or `ButtonRequest` (actually they did and they're entering the passphrase right now).
but that's also a relatively big change

matejcik, 2023-06-19:
also then we have the nasty cornercase of the user entering half the passphrase on host, and then deciding to press the button on Trezor

grdddj, 2023-06-19:
[image]

Currently, we do not have the information about `#1`, so we can only show `HIDDEN WALLET`

grdddj, 2023-06-20:
One UX issue I have - where do we put the second screen? (which has a bad title btw)

If we show it directly after the first one, it has the potential of confusing users (what is the cursor?, some users will probably input the passphrase only via `Suite`).

If we show it every time together with the third one (whole on-device passphrase input), it might become a burden for user to always click this when inputting passphrase.

Very theoretically we might show it _just once_ together with the first passphrase inputting, but that would require allocating some storage space, which I am not sure we want.

[image]

Hannsek, 2023-06-20:
You are right. We don't have to put it anywhere. This screen should have been visible only during the setup of something (pin, recovery). However, passphrase is different. 
Thanks for pointing this out! 

I deleted it from figma.

grdddj, 2023-07-17:
Merged in https://github.com/trezor/trezor-firmware/pull/3135

closed 2023-07-17 by grdddj