trezor/trezor-firmware #3117: Trezor UI freeze during passphrase confirmation

karliatto, 2023-06-29:
**Describe the bug**
Trezor device UI freeze when screen "Next screen will show the passphrase" and click continue if using it with TrezorConnect Explorer and reload TrezorConnect Explorer site before clicking "CONTINUE" in device.

**Firmware version and revision**
Tested with latest master firmware in Trezor emulator and physical device in both the same result.

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: [TrezorConnect  Explorer](https://connect.trezor.io/9/#/method/getAddress) 
 - OS: Linux Ubuntu
 - Browser: Chrome
 - Version [e.g. 90.0.2] Version 114.0.5735.90 (Official Build) (64-bit)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://connect.trezor.io/9/#/method/getAddress with a device that has settings to ask for passphrase
2. Click on "GetAddress"
3. When popup opens click on "Allow once for this session"
4. In popup screen "Export Bitcoin address" click on "Export"
5. When passphrase screen enter a passphrase in the input box and click on "Access Hidden Wallet"
6. Trezor device will display the screen "Next screen will show the passphrase", instead of clicking "CONTINUE" go back to TrezorConnect Explorer tab in the browser and reload it by doing Ctrl-r  for example.
7. After reloading TrezorConnect Explorer go back to device and click "CONTINUE"
8. See error - Trezor device freeze.

**Expected behavior**
Device should not freeze.

**Screenshots**
[image]

**Additional context**
Probably when reloading, host is sending some command to device that is not expected but it should not freeze it anyway.

matejcik, 2023-06-29:
i believe this is because the device sends a `ButtonRequest` to the host and is waiting for a `ButtonAck` that never comes (because you shut down the process that was supposed to send it.

i would expect the same thing to happen on T1, except with black screen instead of "frozen"

also the device should come back alive when you send another command.

matejcik, 2023-06-29:
presumably we should at least fade out before emitting the ButtonRequest (that's what T1 does and I actually have it in the pipeline right now)

worth considering is adding a small timeout for any sort of host response and shutting down when the response doesn't come

romanz, 2026-08-27:
Can we close this issue?

obrusvit, 2026-08-27:
@romanz yes, I don't think we have this problem and we also have "Waiting for connected device" screen. We can reopen as a new issue if something similar starts occurring.
Thanks.

closed 2026-08-27 by obrusvit