trezor/trezor-firmware #3118: Create a mapping of Figma designs to UI tests

grdddj, 2023-06-29:
We want to have an easy way of comparing Figma designs with the current designs in firmware.

Something like:
- `TR recovery flow happy path` = `TR-device_tests-reset_recovery-test_recovery_bip39_t2.py::test_tt_pin_passphrase`
- `TT removing wipe code` = `TT-device_tests-test_msg_change_wipe_code_t2.py::test_set_remove_wipe_code`
- ...

Also, this will uncover designs missing in Figma and/or missing tests for some screens.

matejcik, 2023-07-04:
shouldn't this info be in figma rather than the firmware repo?

grdddj, 2023-07-09:
> shouldn't this info be in figma rather than the firmware repo?

Yes, could be easily in `Figma`/`Notion`, so it is independent of the repository and we can change it without PRs.

As for this issue, this was the natural place to open it, we can move it elsewhere, if it will not get lost/forgotten

grdddj, 2023-07-11:
Incomplete mapping created while checking TR design:

```
Tutorial
- TR-click_tests-test_tutorial.py::test_tutorial_again_and_skip

Wallet creation, Seed backup
- TR-click_tests-test_reset_bip39.py::test_reset_bip39
- TR-click_tests-test_reset_slip39_advanced.py::test_reset_slip39_advanced[2of2]
- TR-click_tests-test_reset_slip39_basic.py::test_reset_slip39_basic[1of1]

Recovery
- TR-click_tests-test_recovery.py::test_recovery_bip39
- TR-click_tests-test_recovery.py::test_recovery_slip39_basic

PIN
- TR-device_tests-test_msg_changepin_t2.py::test_set_pin
- TR-device_tests-test_msg_changepin_t2.py::test_set_failed
- TR-device_tests-test_msg_changepin_t2.py::test_change_pin
- TR-device_tests-test_msg_changepin_t2.py::test_change_invalid_current
- TR-device_tests-test_msg_changepin_t2.py::test_remove_pin
- TR-click_tests-test_pin.py::test_pin_short

Wipe code
- TR-device_tests-test_msg_change_wipe_code_t2.py::test_set_remove_wipe_code
- TR-device_tests-test_msg_change_wipe_code_t2.py::test_set_wipe_code_to_pin

Send BTC
- TR-device_tests-bitcoin-test_signtx_amount_unit.py::test_signtx_testnet[None]
- TR-device_tests-bitcoin-test_signtx_mixed_inputs.py::test_segwit_non_segwit_segwit_inputs

Payment request
- TR-device_tests-bitcoin-test_signtx_payreq.py::test_payment_request[out01]

Send ETH
- TR-device_tests-ethereum-test_signtx.py::test_signtx[Ethereum]
  - !!! shows empty Fee rate and Account !!!
- TR-device_tests-ethereum-test_signtx.py::test_data_streaming
- TR-device_tests-ethereum-test_signtx.py::test_signtx_data_pagination[input_flow_scroll_down]

Receive
- TR-device_tests-bitcoin-test_getaddress.py::test_multisig
- TR-device_tests-bitcoin-test_getaddress_segwit.py::test_show_segwit_altcoin
- TR-device_tests-bitcoin-test_getaddress_show.py::test_show_cancel[m-49h-0h-12h-0-0-InputScriptType.SPENDP2S-4ec777e0
- TR-device_tests-bitcoin-test_getaddress_show.py::test_show_multisig_xpubs[InputScriptType.SPENDP2SHWITNESS--2cf5f03c

Passphrase
- TR-device_tests-test_msg_applysettings.py::test_apply_settings_passphrase
- TR-device_tests-test_msg_applysettings.py::test_apply_settings_passphrase_on_device
- TR-device_tests-test_passphrase_slip39_basic.py::test_2of5_passphrase
- TR-device_tests-test_protection_levels.py::test_passphrase_cached
- TR-device_tests-test_session_id_and_passphrase.py::test_hide_passphrase_from_host
- TR-click_tests-test_passphrase_tr.py::test_passphrase_input[abc 123-mvqzZUb9NaUc62Buk9WCP4L7hunsXFyamT]
- TR-click_tests-test_passphrase_tr.py::test_passphrase_delete

Coinjoin
- TR-device_tests-bitcoin-test_authorize_coinjoin.py::test_get_public_key
- TR-device_tests-bitcoin-test_authorize_coinjoin.py::test_sign_tx_large

Homescreen
- TR-device_tests-test_msg_applysettings.py::test_apply_homescreen_tr_toif_with_long_label
- TR-device_tests-test_msg_applysettings.py::test_apply_homescreen_tr_toif_with_notification

Device name
- TR-device_tests-test_msg_applysettings.py::test_apply_settings

Locktime
- TR-device_tests-bitcoin-test_signtx.py::test_lock_time_blockheight
- TR-device_tests-bitcoin-test_signtx.py::test_lock_time_datetime[2048-08-16 22:14:00]
- TR-device_tests-bitcoin-test_signtx.py::test_lock_time[1-4294967295]

Autolock
- TR-device_tests-test_autolock.py::test_apply_auto_lock_delay

RBF
- TR-device_tests-bitcoin-test_signtx.py::test_information_replacement

FIDO, U2F
- TR-device_tests-webauthn-test_msg_webauthn.py::test_add_remove
- TR-device_tests-webauthn-test_u2f_counter.py::test_u2f_counter

Sign message
- TR-device_tests-bitcoin-test_signmessage.py::test_signmessage[p2pkh long message]

```

closed 2023-07-28 by grdddj