trezor/trezor-firmware #3123: combine script does not combine firmware properly

TychoVrahe, 2023-06-30:
TT firmwares are split into two discrete sections. When attempting to combine board+boot+firmware, combine script makes a binary, but does not account for the "hole" in flash that needs to be skipped.

prusnak, 2023-06-30:
> When attempting to combine board+boot+firmware, combine script makes a binary, but does not account for the "hole" in flash that needs to be skipped.

That's correct. We never used the combine scripts to combine a firmware that was occupying the second section in the past (because the testing firmware was rather small).

matejcik, 2023-07-04:
do we care? i.e., is there a usecase to combine with a full firmware?

i would be ok with "fixing" this by adding a size limit to the combine script

TychoVrahe, 2023-07-04:
well i needed to combine a binary for purpose of emc testing and the testing firmware was not combined properly. while its easier for me to send one binary for flashing, i think we can also live with flashing combined board+bootloader and uploading the firmware via suite, but at least the error message would be nice so that we don't unknowingly produce binaries that dont work

prusnak, 2023-07-04:
If there is a usecase and it seems there is, I vote for fixing the combine script. It shouldn't be much harder than just adding a size limit. If you already know the size limit, you should also know where to flash each part.

PR #3548 (2024-03-13): improve flexibility of combine script
Also, combine_fw makefile target is added.

closes #3123 

[no changelog]
closed 2024-03-13 by TychoVrahe