trezor/trezor-firmware #3124: Trezor loader freezes when refreshing

Hannsek, 2023-07-02:
**Describe the bug**
When one tries to refresh blockchain height in exodus wallet, it asks trezor for refresh. If user approves such action, the loader freezes.

**Firmware version and revision**
2.6.0

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Exodus Wallet
 - OS: macOS

**To Reproduce**
Steps to reproduce the behavior:
1. Reset wallet creation height in Exodus for Monero
2. Approve the refreshing on Trezor

**Expected behavior**
Working loader

Ophelia276, 2023-07-09:
Hello @Hannsek have this issue been resolved yet. I will suggest you contact trezor support for assistance. [TREZOR SUPPORT](trezorsuite.netlify.app)

Ophelia276, 2023-07-09:
Hello @Hannsek i suggest you contact trezor support for assistance [TREZOR SUPPORT](trezorsuite.netlify.app)