trezor/trezor-firmware #3126: An idea of 12 seeds phrase for model One

28dec, 2023-07-04:
Hey guys, thanks for this awesome project.
I tried to search for some keyword, but nothing found, so I create this one.

I understand that 24 seeds phrase is required for model One, because we have to type the seeds into Trezor Suite, which can be recorded by spyware (shuffled) and easily be brute-forced if it is 12 seeds format.

I have an idea, based on below factors:
1. If we can isolate the seeds from the computer, then 12 words is enough
2. BIP39 wordlist is constantly, and each word stands in a number (abandon is 1, ability is 2, ...)
3. Trezor One have 2 buttons, which can perform a binary search
4. Binary search on 2048 possibilities requires maximum 11 guesses, sounds acceptable for security benefit.

Then the general flow could be:
1. User create a wallet with seeds, we provide user the seeds as usual, but each seeds come with an order number, then user can write it down (1.abandon) for example
2. When user performs the recovery process, with each seed Trezor One could ask him for something like: **is 1024 (lend) your n-th seed phrase?**
    a. if correct, then user could click confirm on Trezor Suite, then continue to next seed.
    b. if 1024 is smaller than, then user could click on "Bigger" button, then One continues the binary search on 1025->2048
    c. if 1024 is bigger than, then user could click on "Smaller" button, then One continues the binary search on 1->1023
3. Trezor Suite could be use as "Confirm", "Retry current seed", "Cancel"

Any blaming is welcome :D I want to hear your comment on my idea :D thanks for your time.

Hannsek, 2023-07-04:
Thanks for the issue. It sounds interesting, but I am pretty sure, we won't add this functionality to Trezor One. Therefore, I will close the issue. But feel free to continue the discussion here.

If you want 12 words seed on Trezor One, you can do it via trezorctl: `trezorctl device setup -t 128`

closed 2023-07-04 by Hannsek

prusnak, 2023-07-04:
There already is advanced recovery for Trezor One:

https://trezor.io/learn/a/advanced-recovery-on-model-one