trezor/trezor-firmware #3128: Monero tests can't see WebUSB device

mmilata, 2023-07-04:
Since eacd15383b6801fc09db9d51b31bdfb83da09097 hardware tests in CI are failing. Seems to be locally reproducible:
```
$ trezorctl list
webusb:001:3 - (unnamed) [Trezor T, F569E14AB37A74AC276937FA]

$ ./trezor_monero_tests --trezor-path webusb:
2023-07-04 13:45:00.752	D Device 0 Created
2023-07-04 13:45:00.752	I Test versions Fluorine Fermi' (v0.18.1.0-unknown)
2023-07-04 13:45:00.752	I Testing hardforks [15, 15], sync-test: 1
2023-07-04 13:45:00.753	D Transaction tests for HF 15
2023-07-04 13:45:00.754	D Enumerating Trezor devices...
2023-07-04 13:45:00.755	I [PARSE URI] regex not matched for uri: ^(([^:]*?)://)?(\[(.*)\](:(\d+))?)(.*)?
2023-07-04 13:45:00.755	D Reconnecting...
2023-07-04 13:45:00.757	D Some problems at connect, message: Connection refused
2023-07-04 13:45:00.757	D Failed to connect to 127.0.0.1:21325
2023-07-04 13:45:00.757	E Failed to invoke http request to  /enumerate
2023-07-04 13:45:00.757	E BridgeTransport enumeration failed:Bridge enumeration failed
2023-07-04 13:45:02.260	D Problems at cancel: Bad file descriptor
2023-07-04 13:45:02.260	D Problems at shutdown: Bad file descriptor
2023-07-04 13:45:02.260	D Enumeration yielded 0 Trezor devices
2023-07-04 13:45:02.260	E No matching Trezor device found. Device specifier: "webusb:"
2023-07-04 13:45:02.260	E Could not connect to the device Trezor:webusb:
2023-07-04 13:45:02.261	E gen_trezor_base generation failed: what=Could not connect to the device Trezor:webusb:
2023-07-04 13:45:02.262	D Hardfork added at height: 0, from 1 to 2
2023-07-04 13:45:02.263	E Chain setup failed for HF15
2023-07-04 13:45:02.264	E Exception at [main], what=Couldn't fill transaction sources
2023-07-04 13:45:02.357	D Device 0 Destroyed

```

The test binary can see the device through Trezor Bridge after I start it but it doesn't support debuglink. Emulator seems to also work fine. Any clues @ph4r05?

ph4r05, 2023-07-04:
I will be back 10.7. I would build latest monero, simulator works fine, huge PR waits to be merged. 

Tests are broken on real device, right?

mmilata, 2023-07-04:
Yep, real device.

Btw if the testsuite is basically a patch on top of the monero repo we might be able to easily build it ourselves using nix.

ph4r05, 2023-11-10:
a bit late to the party, but finally: https://github.com/ph4r05/monero/releases/tag/v0.18.1.3-dev-tests-u20.04-01
@mmilata can you pls try this one? Thanks!

mmilata, 2023-12-14:
Thank you! I've updated the URL in #3463 and verified it works with real device, however `gen_trezor_wallet_passphrase` is failing (both device and emulator). Any idea what to look for?

```
2023-12-14 17:01:26.307	E cache_key != m_cache_key. THROW EXCEPTION: error::invalid_password
2023-12-14 17:01:26.307	W /root/monero/src/wallet/wallet2.cpp:4371:N5tools5error16invalid_passwordE: invalid password
2023-12-14 17:01:26.308	E gen_trezor_wallet_passphrase generation failed: what=invalid password
2023-12-14 17:01:26.308	E #TEST# Failed gen_trezor_wallet_passphrase
```

ph4r05, 2023-12-18:
Interesting! I will take a look. There is probably a change causing this. When I created the binary, it worked with master emulator.

ph4r05, 2024-02-04:
I found the problem - just a technicality with setting up a test wallet in the test code, nothing affecting production code. I will release new testing binaries soon / in a few days

ph4r05, 2024-02-04:
https://github.com/ph4r05/monero/releases/tag/v0.18.3.1-dev-tests-u20.04-02

mmilata, 2024-02-21:
Works now, thank you! Though on our self-hosted CI runners I'm getting `Illegal instruction     (core dumped)`, any clue? I'll try to build the test from source to see if it changes anything.

prusnak, 2024-02-21:
> Works now, thank you! Though on our self-hosted CI runners I'm getting `Illegal instruction     (core dumped)`, any clue? I'll try to build the test from source to see if it changes anything.

Sounds like SSE3, SSE4 or similar was available on the host that compiled the binary  and compiler was told to use it. 

We need to fix the compilation via providing the correct march/mcpu flags.

ph4r05, 2024-02-24:
i will try to recompile it again

ph4r05, 2024-02-25:
pls try https://github.com/ph4r05/monero/releases/tag/v0.18.3.1-dev-tests-u18.04-01

mmilata, 2024-02-29:
Thank you, but still the same result. Perhaps it's something to do with our runners, I'll investigate further.

ph4r05, 2024-02-29:
Hmm thats weird. But it can be anything, e.g., some new optimizations in monero build, instructions sets as @prusnak indicated. Can you pls provide `cat /proc/cpuinfo`? Builder machine

```
cat /proc/cpuinfo
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 63
model name	: Intel(R) Xeon(R) CPU E5-2620 v3 @ 2.40GHz
stepping	: 2
microcode	: 0x1
cpu MHz		: 2399.996
cache size	: 16384 KB
physical id	: 0
siblings	: 4
core id		: 0
cpu cores	: 4
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ss ht syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon rep_good nopl xtopology cpuid tsc_known_freq pni pclmulqdq ssse3 fma cx16 pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand hypervisor lahf_lm abm cpuid_fault invpcid_single pti ssbd ibrs ibpb fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid xsaveopt arat umip
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
bogomips	: 4799.99
clflush size	: 64
cache_alignment	: 64
address sizes	: 40 bits physical, 48 bits virtual
power management:
```

ph4r05, 2024-02-29:
Also, can you pls provide more info on the runners?

ph4r05, 2024-03-01:
Btw are you running tests with `MONERO_RANDOMX_UMASK=8` env var? It was related to app crashes (not using it caused crash)

mmilata, 2024-03-01:
We don't. Will try that.

ph4r05, 2024-04-10:
any update @mmilata ?

mmilata, 2024-04-10:
nope, sorry, got stuck working on something else

Hannsek, 2024-04-11:
@ph4r05 
Hey, sorry, but we won't touch this anytime soon as we have higher priorities right now…

matejcik, 2025-02-05:
unclear whether this is actually solved, but we now at least have the new monero binary

PR #3463 (2025-02-05): Update monero tests binary
Fixes #3128. Work in progress, `gen_trezor_wallet_passphrase`.