trezor/trezor-firmware #3136: trezorctl eth web3 error send-tx

nolash, 2023-07-12:
**Describe the bug**

when trying eth send-tx

AttributeError: 'Web3' object has no attribute 'isConnected'. Did you mean: 'is_connected'?

**Firmware version and revision**

$ trezorctl get-features
Features (166 bytes) {
    bootloader_hash: 32 bytes 0xfa12a44fa05fd1d20539358b54f301cee4c3219c9f1bb3a5772ffd609af9e8e2,
    capabilities: [<Capability.Bitcoin: 1>, <Capability.Bitcoin_like: 2>, <Capability.Crypto: 5>, <Capability.Ethereum: 7>, <Capability.NEM: 10>, <Capability.Stellar: 12>, <Capability.U2F: 14>],
    fw_vendor: 'SatoshiLabs',
    initialized: True,
    language: 'en-US',
    major_version: 1,
    minor_version: 11,
    model: '1',
    needs_backup: False,
    no_backup: False,
    passphrase_protection: False,
    patch_version: 2,
    pin_protection: True,
    revision: 20 bytes 0x0d87b55ba4fed7eecc72bf2a94ee473830b095e9,
    safety_checks: <SafetyCheckLevel.Strict: 0>,
    unfinished_backup: False,
    unlocked: False,
    vendor: 'trezor.io',
}

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: [e.g. Trezor Suite, Electrum]

trezorctl

 - OS: [e.g. Windows, macOS, Linux, Android]

linux arch

 - Browser [e.g. Chrome, Firefox, Safari]

na

 - Version [e.g. 90.0.2]

$ pip freeze
aiohttp==3.8.4
aiosignal==1.3.1
async-timeout==4.0.2
attrs==23.1.0
bitarray==2.7.6
certifi==2023.5.7
charset-normalizer==3.2.0
click==8.1.4
construct==2.10.68
construct-classes==0.1.2
cytoolz==0.12.1
ecdsa==0.18.0
eth-abi==4.1.0
eth-account==0.9.0
eth-hash==0.5.2
eth-keyfile==0.6.1
eth-keys==0.4.0
eth-rlp==0.3.0
eth-typing==3.4.0
eth-utils==2.2.0
frozenlist==1.3.3
hexbytes==0.3.1
idna==3.4
jsonschema==4.18.0
jsonschema-specifications==2023.6.1
libusb1==3.0.0
lru-dict==1.2.0
mnemonic==0.20
multidict==6.0.4
parsimonious==0.9.0
protobuf==4.23.4
pycryptodome==3.18.0
referencing==0.29.1
regex==2023.6.3
requests==2.31.0
rlp==3.0.0
rpds-py==0.8.10
simple-rlp==0.1.3
six==1.16.0
toolz==0.12.0
trezor==0.13.7
typing_extensions==4.7.1
urllib3==2.0.3
web3==6.5.0
websockets==11.0.3
yarl==1.9.2

**To Reproduce**

See above

matejcik, 2023-07-12:
looks like web3 did some deprecations while we weren't looking

let's bump the minimum required version of web3 and rename the call I guess?

matejcik, 2023-07-12:
workaround: `pip install "web3<6.0.0"

nolash, 2023-07-12:
thanks.

now i get:

  File "/home/lash/src/home/trezortools/.venv/lib/python3.11/site-packages/parsimonious/expressions.py", line 9, in <module>
    from inspect import getargspec
ImportError: cannot import name 'getargspec' from 'inspect' (/usr/lib/python3.11/inspect.py)

nolash, 2023-07-12:
looks like it works with parsimonious==0.9.0

nolash, 2023-07-12:
```
aiohttp==3.8.4
aiosignal==1.3.1
async-timeout==4.0.2
attrs==23.1.0
base58==2.1.1
bitarray==2.7.6
certifi==2023.5.7
charset-normalizer==3.2.0
click==8.1.4
construct==2.10.68
construct-classes==0.1.2
cytoolz==0.12.1
ecdsa==0.18.0
eth-abi==2.2.0
eth-account==0.5.9
eth-hash==0.5.2
eth-keyfile==0.5.1
eth-keys==0.3.4
eth-rlp==0.2.1
eth-typing==2.3.0
eth-utils==1.9.5
frozenlist==1.3.3
hexbytes==0.3.1
idna==3.4
ipfshttpclient==0.8.0a2
jsonschema==4.18.0
jsonschema-specifications==2023.6.1
libusb1==3.0.0
lru-dict==1.2.0
mnemonic==0.20
multiaddr==0.0.9
multidict==6.0.4
netaddr==0.8.0
parsimonious==0.9.0
protobuf==3.19.5
pycryptodome==3.18.0
referencing==0.29.1
regex==2023.6.3
requests==2.31.0
rlp==2.0.1
rpds-py==0.8.10
simple-rlp==0.1.3
six==1.16.0
toolz==0.12.0
trezor==0.13.7
typing_extensions==4.7.1
urllib3==2.0.3
varint==1.0.2
web3==5.31.4
websockets==9.1
yarl==1.9.2
```

commit 1f45aeb7 (2023-10-19): fixup! fix(python): bump web3 dependency (fixes #3136)
commit e590efad (2023-10-20): fix(python): bump web3 dependency (fixes #3136)
closed 2023-10-20 by matejcik