trezor/trezor-firmware #3153: Rust: `trezor-client`'s build script fails when `protoc` binary is not present

DaniPopes, 2023-07-15:
**Describe the bug**
After the [`0.1.1` release](https://crates.io/crates/trezor-client/versions), [`trezor-client`](https://github.com/trezor/trezor-firmware/tree/b4825f408a5140af43ffcc1744e0df3c5fdf07be/rust/trezor-client) fails building when `protoc` binary is not present due to the addition of the build script.
This makes our CI [workflows fail on `ethers-rs`](https://github.com/gakonst/ethers-rs/actions/runs/5561045481/jobs/10159775784?pr=2510) when building with support for Trezor.

Given that the Rust protobuf bindings are platform agnostic, can their generation be included in the main makefile pipeline like with python's? This would remove the `protoc` dependency and compilation step from its users.

I'd be willing to work on this, if this is desired.

prusnak, 2023-07-15:
> I'd be willing to work on this, if this is desired.

It would be great if you could come up with a build request.

prusnak, 2023-08-15:
@DaniPopes Recently we pushed trezor-client crate v0.1.2. Can you please confirm that the issue is fixed with that release?

prusnak, 2023-08-15:
Ah, my bad. I was under impression that https://github.com/trezor/trezor-firmware/pull/3154 has been merged, but it hasn't.

PR #3154 (2023-09-29): feat(rust): move trezor-client build script to separate crate

Fixes #3153

- Create `trezor-client-build`, which generates the protobuf bindings
  - This is run with `rust/trezor-client/scripts/build_protos` or `make protobuf`, with an optional `--check` flag
- Drive-by: remove Cargo.lock (see [Cargo FAQ](https://doc.rust-lang.org/cargo/faq.html#why-do-binaries-have-cargolock-in-version-control-but-not-libraries))
- Drive-by: build docs.rs documentation with `--cfg docsrs` and add `#[feature(doc_auto_cfg)]` for displaying features alongside items (see [docs.rs](https://docs.rs/about/builds#cr [... cut]
closed 2023-09-29 by prusnak