trezor/trezor-firmware #3162: T2B1 icons in bootloader fix

Hannsek, 2023-07-19:
Fix all icons in bootloader to match the ones used in fw.

grdddj, 2023-07-19:
Can you please provide those that do not match?

Hannsek, 2023-07-19:
there are still bigger arrows, crosses
[image]

`i` icon is not fixed yet
[image]

That is probably it. 🤔

Hannsek, 2023-07-26:
Might be fixed?

Can someone test it? @grdddj

grdddj, 2023-08-07:
Trying to build the bootloader emulator, getting these issues on current master - 59ba113b071bbdd78da43348d9d8d3a4d5ed09c9, even after `make clean`.

A lot of `undefined reference`

Doing `TREZOR_MODEL=R make build_bootloader_emu` - the same also for `TT`

```sh
/nix/store/f4qnwzv6y0nq8lix33jr5ykkyybs6fxf-binutils-2.40/bin/ld: build/bootloader_emu/embed/bootloader/emulator.o: in function `jump_to':
/home/jmusil/trezor-firmware/core/embed/bootloader/emulator.c:70: undefined reference to `STORAGE_AREAS'
/nix/store/f4qnwzv6y0nq8lix33jr5ykkyybs6fxf-binutils-2.40/bin/ld: /home/jmusil/trezor-firmware/core/embed/bootloader/emulator.c:71: undefined reference to `STORAGE_AREAS'
/nix/store/f4qnwzv6y0nq8lix33jr5ykkyybs6fxf-binutils-2.40/bin/ld: build/bootloader_emu/embed/bootloader/emulator.o: in function `main':
/home/jmusil/trezor-firmware/core/embed/bootloader/emulator.c:35: undefined reference to `FIRMWARE_AREA'
/nix/store/f4qnwzv6y0nq8lix33jr5ykkyybs6fxf-binutils-2.40/bin/ld: /home/jmusil/trezor-firmware/core/embed/bootloader/emulator.c:38: undefined reference to `STORAGE_AREAS'
/nix/store/f4qnwzv6y0nq8lix33jr5ykkyybs6fxf-binutils-2.40/bin/ld: /home/jmusil/trezor-firmware/core/embed/bootloader/emulator.c:39: undefined reference to `STORAGE_AREAS'
/nix/store/f4qnwzv6y0nq8lix33jr5ykkyybs6fxf-binutils-2.40/bin/ld: build/bootloader_emu/embed/bootloader/main.o: in function `bootloader_main':
/home/jmusil/trezor-firmware/core/embed/bootloader/main.c:413: undefined reference to `STORAGE_AREAS'
/nix/store/f4qnwzv6y0nq8lix33jr5ykkyybs6fxf-binutils-2.40/bin/ld: /home/jmusil/trezor-firmware/core/embed/bootloader/main.c:521: undefined reference to `FIRMWARE_AREA'
/nix/store/f4qnwzv6y0nq8lix33jr5ykkyybs6fxf-binutils-2.40/bin/ld: /home/jmusil/trezor-firmware/core/embed/bootloader/main.c:325: undefined reference to `FIRMWARE_AREA'
/nix/store/f4qnwzv6y0nq8lix33jr5ykkyybs6fxf-binutils-2.40/bin/ld: build/bootloader_emu/embed/bootloader/messages.o: in function `process_msg_FirmwareUpload':
/home/jmusil/trezor-firmware/core/embed/bootloader/messages.c:594: undefined reference to `STORAGE_AREAS'
/nix/store/f4qnwzv6y0nq8lix33jr5ykkyybs6fxf-binutils-2.40/bin/ld: /home/jmusil/trezor-firmware/core/embed/bootloader/messages.c:597: undefined reference to `FIRMWARE_AREA'
/nix/store/f4qnwzv6y0nq8lix33jr5ykkyybs6fxf-binutils-2.40/bin/ld: /home/jmusil/trezor-firmware/core/embed/bootloader/messages.c:622: undefined reference to `FIRMWARE_AREA'
/nix/store/f4qnwzv6y0nq8lix33jr5ykkyybs6fxf-binutils-2.40/bin/ld: /home/jmusil/trezor-firmware/core/embed/bootloader/messages.c:655: undefined reference to `FIRMWARE_AREA'
/nix/store/f4qnwzv6y0nq8lix33jr5ykkyybs6fxf-binutils-2.40/bin/ld: build/bootloader_emu/embed/bootloader/messages.o: in function `bootloader_WipeDevice':
/home/jmusil/trezor-firmware/core/embed/bootloader/messages.c:685: undefined reference to `WIPE_AREA'
```

@TychoVrahe surely you will know what happened there

matejcik, 2023-08-07:
i think you'll need to rebase on #3185 ?

TychoVrahe, 2023-08-07:
yup there is a fix in the prodtest PR in core/SConscript.bootloader_emu . I keep breaking this emu build, we should add it to CI.

grdddj, 2023-08-07:
Thanks, the rebase on that branch helped, I just had to do this one fix, there were mismatched return types:
```
-- a/core/embed/trezorhal/unix/secret.c
+++ b/core/embed/trezorhal/unix/secret.c
@@ -2,7 +2,7 @@
 
 void secret_write(uint8_t* data, uint32_t offset, uint32_t len) {}
 
-void secret_read(uint8_t* data, uint32_t offset, uint32_t len) {}
+secbool secret_read(uint8_t* data, uint32_t offset, uint32_t len) { return secfalse; }
```

Both the icons seem to be fine now, see the screens below:

[image]
[image]

@Hannsek please check and probably close this

> we should add it to CI.

I will create an issue for this

Hannsek, 2023-08-09:
Seems already resolved.

closed 2023-08-09 by Hannsek