trezor/trezor-firmware #3183: Shamir info button not working correctly

grdddj, 2023-07-27:
In `TT` Shamir recovery flow, when clicking the info button on the recovery homescreen and then going back, the screen is blank

e.g. https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/4751858982/artifacts/test_ui_report/passed/TT-device_tests-reset_recovery-test_recovery_slip39_advanced.py%3A%3Atest_secret_click_info_button%5Bshares0-c2d2-850ffa77.html (below)

[image]

conduition, 2023-08-15:
Replicated by doing:

```
trezorctl device setup -b advanced -t 128
```
And selecting two groups with one share each, with a two-group threshold. Then recovering and hitting the more info button here.

[image]

After pressing 'more info':

[image]

After pressing 'continue':

[image]

```
trezorctl wipe
trezorctl recover
```

I replicated the bug, but it appears slightly differently on my emulator from the screenshots in OP. 

I'm investigating the cause now

Hannsek, 2023-08-16:
I suppose you have an unreleased version of fw, 2.6.1, right?

conduition, 2023-08-17:
@Hannsek That's correct. I replicated on a41a8c5f8721f50d89a8f8b9efd9eb3a3809854c ([tree](https://github.com/trezor/trezor-firmware/tree/a41a8c5f8721f50d89a8f8b9efd9eb3a3809854c))

closed 2023-08-23 by grdddj