trezor/trezor-firmware #3210: Chunked addresses

Hannsek, 2023-08-15:
From Suite repo:

> As the user would be able to change the view of the addresses (chunked/not-chunked), Suite should remember the last state. So if the user chose to have it chunked last time, next time he/she receives/sends, he/she should see the address chunked, and vice versa.
> 
> Each view should have its own message which would be sent to Trezor. According to this message, Trezor would display chunked/not-chunked address.

Is this possible? Receiving messages from suite and change the view appropriately? @matejcik @grdddj @mmilata 

[T2B1](https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Firmware?type=design&node-id=3790-11545&mode=design&t=SCxX0NadL7vcFa04-0) 

[T2T1](https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Firmware?type=design&node-id=478-3&mode=design&t=GVbDeO39c8DE3E39-0)

grdddj, 2023-08-15:
> Is this possible? Receiving messages from suite and change the view appropriately?

I guess technically possible, but probably not easy to do. The details of how to display something are very far away from receiving a protobuf message from `Suite`.

Seems we would need to have a `single global layout` so we can check it is the address screen and force it to recreate/update with changed parameters?

prusnak, 2023-08-15:
Easiest is to expand the `GetAddress` protobuf message with field `optional chunked bool` and show chunked address only if it is set to true.

matejcik, 2023-08-15:
> Is this possible? Receiving messages from suite and change the view appropriately? @matejcik @grdddj @mmilata

it should be, with a tiny amount of trickery.
`ctx.wait(layout)` is shortcut for `loop.race(layout, ctx.read_any())`

if instead of `ctx.wait(layout)`, we use explicit `loop.race(layout, ctx.read(GetAddress))`, we will be able to intercept a subsequent GetAddress without ever leaving the running workflow.

the only difficulty is that the `ctx.wait` call is hidden deep enough in the stack that we'll need a way to propagate it out. we might want to move `show_address` out of layouts and into `apps.common` and implement the logic there, take an argument specifying what message to wait for...

closed 2023-09-15 by grdddj