trezor/trezor-firmware #3238: Emulator does not display properly on macOS

guoyk93, 2023-08-27:
**Describe the bug**
The emulator `emu.py` does not display properly on macOS

**Firmware version and revision**
`master` at `6c97dfef78188a994c33f2c411b9cbd7b3c1b97d`

**Desktop/smartphone setup (please complete the following information):**
 - OS: macOS
 - Device: M1 MacBook Air

**To Reproduce**
Simplely follow the documentation to run emulator

**Expected behavior**
It should display properly

**Screenshots**
[image]

And the touch area is also not correct, it is mapped to somewhere around the red rectangle.

[image]

Hannsek, 2023-08-28:
I don't have that problem on mac pro M1, Ventura 13.5

[image]

@trezor/qa do you have this problem as well?

Hannsek, 2023-08-28:
I don't have this issue on Mac pro M1, Ventura 13.5 on both, built locally or from trezor-user-env emulators
[image]

guoyk93, 2023-08-28:
@prusnak 

MacBook Air M1 2020 with macOS Ventura 13.5

I installed `sdl2` from `brew`, as described at https://docs.trezor.io/trezor-firmware/core/build/emulator.html

I see an `X.org` icon from @Hannsek 's screenshot. I just tried installing `XQuarz`, still not work.

guoyk93, 2023-08-29:
@prusnak @Hannsek I tried using `nix-shell`, and it's ok this time !!!

The previous problem may be caused by `brew`, since it always install the latest version. Maybe some version of libraries are not compatible with the code.

Thanks for all.

---

And I'm thinking, maybe we can make the `Emulator` part of the documentation more simple and understandable for beginners, even pack it to an application.

I'm tring migrate my crypto assets from old mnemonic words to new ones (since the old ones have protential risk of leaking), and finding out that BIP39 with extra passphrase (Hidden wallet) are actually not widely adopted by other software wallets.

Since I got only one `Trezor T` for now (for new mnemonic words) and I have to do this quickly, I have to run a `Trezor` emulator to access the hidden wallets, and trasfer assets. 

Maybe some people who don't have software development knowledge also have this demand.

prusnak, 2023-08-29:
Reopened because the issue hasn't been investigated and fixed yet.

guoyk93, 2023-08-31:
@prusnak @Hannsek

I investigated this bug a litter more.

It only occurs when using `brew` to prepare the environment and using the built-in retina display of MacBook, external display is fine.

Environment prepared by `nix-shell` is ok with the built-in retina display of MacBook.

**SDL versions installed from `brew`:**

- `sdl2 (2.28.2)`
- `sdl2_image (2.6.3)`

guoyk93, 2023-08-31:
It should be just a minor issue