trezor/trezor-firmware #3241: Rewrite `build-docker.sh` in Python

matejcik, 2023-08-30:
It has grown too complex and it's gonna get worse before it gets better.

Desired features:
* build from specified branch, or currently checked-out branch by default.
  * fetching a different repository seems not useful
* specify on command line which artifacts to build (as opposed to which to skip)
  * instead of core/legacy, base the artifact names on internal model names
  * allow `--all`, also allow "all of T2T1", also allow "T2T1 btc only, T2T1 prodtest, T2B1 bootloader"
    * figure out reasonable naming of the artifacts
    * allow something like "-dev" for `PRODUCTION=0` builds
* emit and keep the "build-<artifact-name>.sh" in `build/`

optionally, instead of building, emit a shell script for reproducible-build-by-anyone-at-any-time:
   1. (maybe optionally clone the whole repo??)
   2. checkout the given tag/branch by explicit commit hash
   3. generate the "build-<artifact-name>.sh" as a heredoc
   4. execute docker with the right arguments to produce that one artifact
   5. optionally check results against explicit values?

the result of this last feature should be useful for build attestation, see #2200

prusnak, 2023-08-30:
It would be great if the script had no external dependencies, only Python standard library.

matejcik, 2023-08-30:
i'm actually wondering whether the script should allow using non-current branches? history shows that the script itself affects build results, so running from branch A might not give reproducible result from branch B

maybe the branch should not be an argument, and instead the script should warn if the checkout is dirty

in that case directly importing code from the local branch would be acceptable (but not that useful because even the imported code is only allowed stdlib + direct imports)

on a system with git + python3 + docker, the following must work:
```
git clone https://...
cd trezor-firmware
git checkout release/v3.0.0
python3 build-docker.sh
```

prusnak, 2023-08-30:
> maybe the branch should not be an argument, and instead the script should warn if the checkout is dirty

+1

> on a system with git + python3 + docker, the following must work:

+ 1