trezor/trezor-firmware #3244: Small delay on startup when other then north orientation of the screen is set.

Hannsek, 2023-09-04:
**Describe the bug**
When display orientation is not set to North, the loading screen with the lock and Trezor Model T on startup just blinks.

**Firmware version and revision**
TT 2.6.0

**To Reproduce**
Steps to reproduce the behavior:
1. set display orientation to something else then North
2. connect trezor to power
3. see the glimpse of the screen

**Expected behavior**
The screen (with logo and text) should stay longer on startup 

**Screenshots**
https://satoshilabs.slack.com/archives/D03Q8LTUEMU/p1693397867936819

Hannsek, 2023-09-04:
@matejcik does this seem to be the low hanging fruit?

matejcik, 2023-09-04:
hard to say. I don't know what is the cause so :woman_shrugging:

closed 2024-01-09 by TychoVrahe