trezor/trezor-firmware #3246: ETH fees units

Hannsek, 2023-09-04:
Please change ETH fees units in ETH send flow as follows:
[image]
[image]

Maximum fee: wei ETH -> ETH
Gas price: ETH -> Gwei

Add (if not already satisfied) priority fee key.

Hannsek, 2023-11-22:
@grdddj please take a look, it should be low hanging.

grdddj, 2023-11-22:
> @grdddj please take a look, it should be low hanging.

Right, PR done - https://github.com/trezor/trezor-firmware/pull/3414

It was quite easy to change, but I am not sure we want it like this --- please check the results/diffs

closed 2023-12-01 by matejcik