trezor/trezor-firmware #3253: Can't sign typedata in modal one since MM version 10.31 update

parmesh-shiroya, 2023-09-06:
**Describe the bug**
I can't sign any type data message from MM version 10.31 it works fine on MM 10.30.
When signing type data. MM open Trezor web page where it ask to export a private key. Clicking on it throws back to site with MM message 'unknown error'. 
This is not happening on Trezor Modal T. Only issue with Modal One

**Firmware version and revision**
10.12.1

**Desktop/smartphone setup (please complete the following information):**
 - Metamask
 - Mac, Windows
 - Brave, Chrome
 - 1.52.129

**To Reproduce**
Steps to reproduce the behavior:
we have created site to test these behaviour. You can use that or just try to list any item on any marketplace. Or call eth_signTypedData_v4 method 
1. Go to ['this site'](https://seaportsignbulkorders.ryanio.repl.co/)
2. Enter 1 in Number of orders to create
3. Click on 'create order'
4. connect Trezor Modal one through metamask. And check the console.

**Expected behavior**
It should sign a message.

matejcik, 2023-09-06:
duplicate of https://github.com/trezor/trezor-suite/issues/9365 (and wrong repository too!)

matejcik, 2023-09-06:
hint:

> I can't sign any type data message from MM version 10.31 it works fine on MM 10.30.

This **strongly indicates** that the problem is not with Trezor firmware. The right thing to do here would have been to report to Metamask (who would then investigate and ping the problem onto Trezor Connect)

closed 2023-09-06 by matejcik