trezor/trezor-firmware #3331: Model One reporting register when asked to authenticate

bosomt, 2023-10-05:
**Describe the bug**

Model One reporting register when asked to authenticate.
I will attach USB capture soon.

See https://satoshilabs.slack.com/archives/CL3U40DPA/p1696492537354129?thread_ts=1696341444.303159&cid=CL3U40DPA

**Info:**
 - Suite version: desktop 23.10.0 (0a29349b790372765381f587de45d2b957d2b94e)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuiteDev/23.10.0 Chrome/112.0.5615.204 Electron/24.5.1 Safari/537.36
 - OS: MacIntel
 - Screen: 1440x900
 - Device: Trezor T1B1 1.12.1 regular (revision 1eb0eb9d91b092e571aac63db4ebff2a07fd8a1f)
 - Transport: BridgeTransport 2.0.32

**To Reproduce**
Steps to reproduce the behavior:
1. Connect model One device with Universal firmware
2. Try to login to website with U2F enabled
3. Compare device dialogue with website dialogue

**Screenshots**
[image]

bosomt, 2023-10-19:
@andrewkozlik attaching capture from macos 
[T1_mac_github_login.pcapng.zip](https://github.com/trezor/trezor-firmware/files/13039287/T1_mac_github_login.pcapng.zip)

andrewkozlik, 2024-01-05:
I analyzed the log and Trezor is working as expected. However, what the computer is doing makes no sense. First, it's trying to authenticate using a FIDO2 application ID for **GitHub**, seriously? It also presents a FIDO2 credential that was generated using the FIDO2 protocol, obviously using a Trezor T, because Model 1 doesn't support FIDO2. Second, the computer is trying to register using some unknown application ID e84541eaf207f7d75ad051434770f6d1a9bf62f7ea9be514fd4e0ca8272b1deb. I wonder what this ID represents. It could be a dummy ID used on Macs, in which case we should add it to our list of bogus app IDs, so that we can display "Not registered". But I was unable to find this ID mentioned anywhere, so who knows.

Conclusion: It's a Mac bug.

Details for the record:

First there is a U2F_AUTHENTICATE request, with parameters:
control byte: check-only
challenge: 3e38cf988e7bd7a866d3e751b44ef8da5dc53f72c650215ee5d51eb144a283a6
application: github.com (3aeb002460381c6f258e8395d3026f571f0d9a76488dcd837639b13aed316560)
key handle: f1d00200e0dc844be35866a77097253cd372021ec071326911ba519eaa85e3790ef40e62269434f2a2d6e5d684072a42431d5b3574e9b7c9035e4bb1f9e1d5614ad86e83afb18c2dced8e030b6671dcefb55e85702b3b1dddd5f5b80086d26a5993fb58546e64a0d2e3a0a7e6a6731f6c9cadb525731838a8852ec895d5f1c29

Second there is a U2F_REGISTER request, with parameters:
P1: 0x03
challenge:  d0cee6fc7dbf599a919db8fb951311269f0eb781f7841c6cc0544ad9da34154b
application: unknown (e84541eaf207f7d75ad051434770f6d1a9bf62f7ea9be514fd4e0ca8272b1deb)
Note that the U2F spec doesn't allow anything other than 0x00 for P1. This may allegedly be for compatibility with some older devices. We ignore it anyway.

closed 2026-08-19 by andrewkozlik