trezor/trezor-firmware #3375: impossible to SHOW a whole long passphrase in TS3

02curls, 2023-10-27:
**Describe the bug**
When entering a LONG passphrase ON DEVICE (TS3), it is impossible to check the whole entered passphrase on device by clicking SHOW as there is no way to "scroll" left towards the beginning of the long passphrase, only the end of the passphrase is currently shown

**Firmware version and revision**
2.6.2. btc only

sophiadapers88, 2023-10-31:
[image]
my MM - Trezor skipping this step. Now I cannot access my wallet address in MM

brianddk, 2024-02-19:
> my MM - Trezor skipping this step. Now I cannot access my wallet address in MM

I'm not sure MM had Trezor-3 support back in October.

brianddk, 2024-02-19:
> When entering a LONG passphrase 

Second sighting reported on by @spatafore on Reddit with FW v2.6.4-btc-only