trezor/trezor-firmware #3378: Noise over USB Bus causes touchscreen issues

cryptosig, 2023-10-31:
Touchscreen stops working as expected, unlocking is hard to press accurately and during signing transaction the progress bar continually resets and never completes 

**Firmware version and revision**
Describe firmware version and revision you are using when encountering the bug.
Version 2.5.3
**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: [e.g. Trezor Suite, Electrum]
 - OS: [e.g. Windows, macOS, Linux, Android]
 - Browser [e.g. Chrome, Firefox, Safari]
 - Version [e.g. 90.0.2]
Trezor Model T, Trezor Suite, Ubuntu Linux
**To Reproduce**
Steps to reproduce the behavior:
Use usb devices and conduct a session

**Additional context**
I have had these problems with my Trezor T for awhile, on this particular occasion it took many many attempts to even unlock because the touchscreen was not able to register the touches properly and consistently. I was conducting a transaction and unable to complete the transaction signing due to the "hold to sign" not working properly as it was jittering back to 0% progress. 

I was able to track down the problem and was able to temporarily fix the issue. Chrome was open in the background, though I was using Trezor Suite to conduct the transaction, youtube was playing audio in the background, I have laptop with USB audio speakers as they are common in newer laptops, I stopped the audio from playing and the trezor device was no longer getting USB interference. The touchcreen worked again after stopping the usb interference.

Hannsek, 2023-10-31:
Please update to the newest version of fw and see if it happens there as well.

matejcik, 2023-10-31:
This is very clearly a hardware issue so from the outset I'm closing this as there is literally nothing we can do about it.

With that said:
There is no such thing as USB crosstalk, unless you have an extremely-broken-to-the-point-of-dangerous USB hub.

What is happening here is _electrical_ interference. If stopping the audio helped, that means that the main suspect are the speakers, sending some sort of high frequency electrical noise over your wires.

It's pretty common that, e.g., when you have two devices plugged into mains (e.g., charging your laptop + separately powering the speakers), there is electrical noise traveling over the wires. If this is your configuration, see if unplugging your laptop from the charger helps.
If the speakers are powered from USB, I'm afraid that you'll need to get yourself:
* a powered hub that can should isolate them electrically from the rest of the system
* or just less crappy speakers.
* or, if the problem is power draw, maybe just a stronger charger for the laptop would help? (does the problem also happen while on battery?)
In similar cases in the past, it turned out that it's _not just Trezor_ after all -- maybe your mouse lags a little, maybe you're getting bad speeds from your flash drives.

...that, or just pause your music when you need to confirm a transaction....

closed 2023-10-31 by matejcik