trezor/trezor-firmware #3411: Fix invalid encoding of Optiga signatures

matejcik, 2023-11-22:
The Optiga chip sometimes seem to produce invalid DER encoding of ECDSA signatures: where DER specifies that INTEGERs must be prefixed by 0x00 if and only if the leading byte is >= 0x80, Optiga seems to always(?) add the 0x00 prefix. The resulting DER encoding is invalid -- but can be easily fixed by stripping the wrong 0x00 bytes.

We need to fix this in two places:
* `optiga_calc_sign()` when producing the signature, so that all results (incl. e.g. in `AuthenticateDevice`) already come out correct, and
* when providing stored device certificate, which might have been signed by a buggy optiga too

prusnak, 2023-11-23:
> Optiga seems to always(?) add the 0x00 prefix.

So what exactly is the nature of the issue? If Optiga always added 0x00 prefix, then the probability would be 50%, while comment introduced in https://github.com/trezor/trezor-suite/pull/10033 says the probability is 1/256.

andrewkozlik, 2023-11-23:
Optiga correctly adds the 0x00 prefix byte when necessary. The problem is that it always makes the contents octets of the integers at least 32 bytes long (in case of P-256). So that way you can for example get this invalid DER encoding of an integer:
`0220000102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f`
which should be
`02200102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f`.
In other words the hexadecimal encoding of the invalid integers always matches the regexp `^022000[0-7]`.

The likelihood of the 32nd least significant byte being `0x00` and at the same time the 31st least significant byte being `< 0x80` by change is approximately 1 in 512. The probability of this happening in at least one of the two integers that make up the signature is approximately 1 in 256.

onvej-sl, 2023-11-28:
> which should be
02200102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f..

I think, it should be 02**1f**0102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f.

PR #3417 (2023-11-30): Fix invalid encoding of signatures from Optiga
Fixes #3411. Tested on development device using 8 `AuthenticateDevice` calls. There is no guarantee that any of the signatures were corrupt, but there is a unit test in `crypto`, which should check that integers get reencoded correctly.
closed 2023-11-30 by andrewkozlik