trezor/trezor-firmware #3418: Missing documentation on how to verify T2B1

maaft, 2023-11-25:
In [docs](https://docs.trezor.io/trezor-firmware/common/reproducible-build.html) you can find instructions  how to verify self-build firmware for model T and model One.

How can this be done with the new Safe 3?

matejcik, 2023-11-25:
the Safe 3 firmware is the same type and format as the TT so the exact same instructions apply.

the only change (which we do need to document) is that the path to the TS3 binary is `build/core-r/fimware/firmware.bin`, and similarly for the model T the path changes to `core-t`

Giszmo, 2024-03-27:
It's `core-R` and `core-T` respectively.

bosomt, 2024-10-14:
baybe we can consider adding instructions for macOS ?
Maybe command are need to be started in nix-shell ?
Results from my `macARM 15.0.1 (24A348)`

```
❯ bash build-docker.sh core/v2.8.3
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 2563k  100 2563k    0     0  7760k      0 --:--:-- --:--:-- --:--:-- 7768k
usage: sha256sum [-bctwz] [files ...]
❯ cd build/core
cd: no such file or directory: build/core
❯ cd build
cd: no such file or directory: build
```

Hannsek, 2024-10-14:
@bosomt feel free to go ahead 😄

commit cc84c097 (2024-10-11): docs: update reproducibility instructions

fixes #3418
commit d5f2ea48 (2024-10-11): docs: update reproducibility instructions

fixes #3418
PR #4255 (2024-10-29): docs: update reproducibility instructions
fixes #3418
fixes #4254 

@bosomt please try reproducible builds based on these instructions
@Hannsek please review in terms of readability
commit ca1ed7ce (2024-10-29): docs: update reproducibility instructions

fixes #3418
closed 2024-10-29 by matejcik