trezor/trezor-firmware #3460: cannot claim aqua airdrop on trezor

rednek123, 2023-12-13:
when trying to claim aqua on stellar network via aqua.network and wanting to sign in stellar labs. i get an error.

when i click claimable balance and want to sign in stellar labs via trezor connect.

i first initialze signing with account. press confirm
then confirm timebounds
then no memo set (i click the green "v"
it get stuck on the next screen. (NO memo set!)

the labaratory then give me the error

Error:"Cannot destructure property 'type' of 't[s]' as it is undefined."

Firmware version 2.6.3

Desktop Windows 10 Pro
 - Wallet software: Trezor model t
 - OS: Windows
 - Browser Chrome
 - VersionVersie 120.0.6099.71 (Officiële build) (64-bits)

[image]
[image]
[image]
[image]
![a5](https://github.com/trezor/trezor-firmware/assets/153725511/e31f8262-be74-4090-a2e1-aa1ae33abe34
I

@overcat 

i hope this is enoug informatioon to get a good image of the problem.
and i hope someone can squash the bug ,-)

matejcik, 2023-12-13:
> the labaratory then give me the error
> 
> Error:"Cannot destructure property 'type' of 't[s]' as it is undefined."

this makes it not a Trezor firmware error. is it a Connect problem (if yes we can move it there) or Laboratory problem?

(also `ClaimClaimable` command is in the firmware repo but not available in any released Trezor firmware yet)

Hannsek, 2024-01-10:
@mroz22 is this a Connect issue?

mroz22, 2024-01-10:
of course it has nothing to do with firmware