trezor/trezor-firmware #3471: Stellar: add support for Soroban Smart Contract.

overcat, 2023-12-21:
Stellar will upgrade its mainnet to [Protocol 20](https://stellar.org/blog/developers/protocol-20-upgrade-guide) on January 30, 2024. This marks the most significant upgrade since the launch of the Stellar network, introducing support for Soroban Smart Contracts. I hope that we can add support for it in Trezor.

**Spec:**
We need to perform the following tasks:

- Add support for `InvokeHostFunctionOp` / `ExtendFootprintTTLOp` / `RestoreFootprintOp`. (See [XDR definition](https://github.com/stellar/stellar-xdr/blob/bb54e505f814386a3f45172e0b7e95b7badbe969/Stellar-transaction.x#L668-L673))
- Add support for signing Soroban Authorized Invocation. (See [authorization docs](https://soroban.stellar.org/docs/fundamentals-and-concepts/authorization))

The complete documentation can be found [here](https://soroban.stellar.org/docs), and the complete XDR definition can be found [here](https://github.com/stellar/stellar-xdr/tree/bb54e505f814386a3f45172e0b7e95b7badbe969).

**Design:**
Furthermore, I have already validated this locally. Of course, this is not the final version, and I will make some optimizations. After completing these optimizations, I will submit a new pull request. Below is a demonstration:

https://github.com/trezor/trezor-firmware/assets/2368403/cf0a6ede-3a84-4fa3-b041-0a50a5b9e085

https://github.com/trezor/trezor-firmware/assets/2368403/0e3c1b1a-ef8f-41b8-bd7f-1ac1b20152a6

matejcik, 2023-12-21:
once again i have to give kudos to Stellar for their schema being extremely sanely designed :100: 

with that said, please hold off on this a little. if you already have proof of concept, you should be able to measure how much flash space the implementation eats (by building firmware with and without it)

overcat, 2023-12-26:
> how much flash space the implementation eats

```python tools/size/apps.py```

master branch
```
    56_124: bitcoin              (  439 symbols)
    48_758: monero               (  349 symbols)
    35_606: cardano              (  301 symbols)
    26_307: solana               (   81 symbols)
    22_512: webauthn             (  182 symbols)
    20_666: common               (  161 symbols)
    16_976: management           (  100 symbols)
    14_891: ethereum             (   83 symbols)
    10_627: nem                  (   77 symbols)
**   8_548: stellar              (   61 symbols) ** 
     6_997: eos                  (   53 symbols)
     6_543: tezos                (   34 symbols)
     4_949: zcash                (   52 symbols)
     3_831: misc                 (   19 symbols)
     2_773: binance              (   16 symbols)
     2_713: ripple               (   22 symbols)
       723: debug                (    3 symbols)
       715: homescreen           (    6 symbols)
SUMMARY: 18 categories, 2_039 symbols, 290_259 bytes in total.
```

the poc branch
```
    56_124: bitcoin              (  439 symbols)
    48_758: monero               (  349 symbols)
    35_606: cardano              (  301 symbols)
    26_307: solana               (   81 symbols)
    22_512: webauthn             (  182 symbols)
    20_666: common               (  161 symbols)
    16_976: management           (  100 symbols)
**  15_773: stellar              (   95 symbols) **
    14_891: ethereum             (   83 symbols)
    10_627: nem                  (   77 symbols)
     6_997: eos                  (   53 symbols)
     6_543: tezos                (   34 symbols)
     4_949: zcash                (   52 symbols)
     3_831: misc                 (   19 symbols)
     2_773: binance              (   16 symbols)
     2_713: ripple               (   22 symbols)
       723: debug                (    3 symbols)
       715: homescreen           (    6 symbols)
SUMMARY: 18 categories, 2_073 symbols, 297_484 bytes in total.
```

@matejcik, currently it consumes 7225 bytes, but with optimization it should be able to reduce some, although I don't think it will be much.

overcat, 2024-01-03:
I added the compare.sh information to the PoC PR(#3479).

overcat, 2026-01-21:
A quick note on the UI: Currently, it shows a memo warning, but since Soroban transactions don't actually support memos, this can be a bit confusing. I’m thinking of optimizing the UI so that this reminder only pops up for standard payment operations.