trezor/trezor-firmware #3473: Update Bootloader changelog for Trezor T on trezor.io/learn

CodeShakingSheep, 2023-12-26:
Released bootloader versions

- 2.1.1 - June 2023
- 2.1.2 - August 2023
- 2.1.3 - September 2023
- 2.1.4 - November 2023

for Trezor T are not mentioned on https://trezor.io/learn/a/bootloader-changelog in the version overview. Would love to see them added there.

matejcik, 2024-01-02:
I have passed the report to the right people. Closing the issue here, as the firmware repo can't do anything about it, the `CHANGELOG.md` file in the repo is in order.

closed 2024-01-02 by matejcik

CodeShakingSheep, 2024-01-02:
Thanks for your reply @matejcik. Where would have been the right place to address this? I couldn't find a repository for the website.

Hannsek, 2024-01-02:
The best way is to use Trezor support.