trezor/trezor-firmware #3497: Send Ethereum transaction for sign

kiruxaspb, 2024-01-19:
**Describe the bug**
Problem with send transaction to Trezor for sign transaction

**Firmware version and revision**
trezor lib for Python
'trezor==0.13.8'

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Python 3.11.6
 - OS: MacOS

**To Reproduce**
I want send ethereum transaction to Trezor One for signature, but i get errors with conversion types.

**Code**
```
"value": 0,
"gas": 46541,
"from": "0x2Fc6bd8e2540db7247A0772aA7eDBFA0A59d78C0",
"nonce": 205,
"gasPrice": 280,
"chainId": 5,
"to": "0xd24292864063d0DA1F294AC65D74d55a44F4766C",
"data": "0x095ea7b30000000000000000000000006245f856eddbdb3ed6a3c64385b27a78b42f65e10000000000000000000000000000000000000000000000000000000005f5e111"
```
```python
self.client = TrezorClient(transport=get_transport(), ui=ui.ClickUI())
        self.index = index
        self.path = tools.parse_path(f"44'/60'/0'/{self.index}")
        self.account = ethereum.get_address(self.client, self.path)
```

```python
signature = ethereum.sign_tx(
            client=self.client,
            n=self.path,
            nonce=transaction["nonce"],
            gas_price=transaction["gasPrice"],
            gas_limit=transaction["gas"],
            chain_id=transaction["chainId"],
            to=bytearray.fromhex(transaction["to"][2:]),
            data=bytearray.fromhex(transaction["data"][2:]),
            value=int(transaction["value"])
        )
```

```
File "/Users/Work/venv/lib/python3.11/site-packages/trezorlib/tools.py", line 308, in wrapped_f
    return f(client, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/Work/venv/lib/python3.11/site-packages/trezorlib/ethereum.py", line 229, in sign_tx
    response = client.call(msg)
               ^^^^^^^^^^^^^^^^
  File "/Users/Work/venv/lib/python3.11/site-packages/trezorlib/tools.py", line 308, in wrapped_f
    return f(client, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/Work/venv/lib/python3.11/site-packages/trezorlib/client.py", line 257, in call
    resp = self.call_raw(msg)
           ^^^^^^^^^^^^^^^^^^
  File "/Users/Work/venv/lib/python3.11/site-packages/trezorlib/client.py", line 154, in call_raw
    self._raw_write(msg)
  File "/Users/Work/venv/lib/python3.11/site-packages/trezorlib/client.py", line 163, in _raw_write
    msg_type, msg_bytes = self.mapping.encode(msg)
                          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/Work/venv/lib/python3.11/site-packages/trezorlib/mapping.py", line 64, in encode
    protobuf.dump_message(buf, msg)
  File "/Users/Work/venv/lib/python3.11/site-packages/trezorlib/protobuf.py", line 491, in dump_message
    assert isinstance(svalue, str)
AssertionError
```

According to the value type, the transmitted value is exactly int
Also i trying pass the value in n: self.account, but it ends up with a similar error

How i can fix this?

matejcik, 2024-01-19:
this is not a support site. next time please direct your inquiries either to the Trezor Forum at https://forum.trezor.io/ or to the Gitter trezor-community.

that said, `to` should be the text `"0xabcd0123..."`, _not_ conversion to bytes

matejcik, 2024-01-19:
(fwiw if you used VScode or PyCharm with type checking enabled, you would get squiggly lines when you do things like this)

closed 2024-01-19 by matejcik