trezor/trezor-firmware #3499: [Model T] "Definition out of date" error on one particular ETH contract address

the-emerald, 2024-01-23:
**Describe the bug**
Device returns "Definition out of date" when attempting to call the contract `0x7b35Ce522CB72e4077BaeB96Cb923A5529764a00` on Ethereum Mainnet. This happened across two different Model Ts with two different devices running different OSes.

**Firmware version and revision**
Describe firmware version and revision you are using when encountering the bug.

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: MetaMask and Frame
 - OS: Windows, Mac and Linux
 - Browser Firefox and Chrome

**To Reproduce**
Steps to reproduce the behavior:
1. Create a transaction that calls `0x7b35Ce522CB72e4077BaeB96Cb923A5529764a00`. Any function will do.
2. Attempt to sign the transaction. Trezor will return "Definition out of date".

**Expected behavior**
The transaction to go through.

**Screenshots**
N/A

**Additional context**
N/A

matejcik, 2024-01-23:
Per the definition data, the contract is "Impermax OLD", neither ethereum-lists/tokens nor coingecko knows about it, so the "definition is outdated" message is _technically correct_ (the best kind of correct).

Where do you people keep coming up with all this old junk??

matejcik, 2024-01-23:
nevertheless.
we deleted the definition file, so next time you try this, you should get the correct thing (contract address confirmation screen and UNKN ticker symbol)

closed 2024-01-23 by matejcik