trezor/trezor-firmware #3503: Easy to get "stuck" in recovery mode

matejcik, 2024-01-26:
On TT, when the user gets to the "select number of words" screen, there is no Cancel button.
By that time, the recovery mode is already on, and if you unplug and replug your Trezor, you jump straight back into "select number of words".

The thing the user can do is select a number to move on into the recovery homescreen, which has a Cancel button.

Solutions:

* add a Cancel button to the "select number of words" screen -- very slightly nonpreferred because that's a design change
* do not go to recovery mode before number is selected (so if the user unplugs at this screen, they go back to normal Trezor homescreen)

matejcik, 2024-01-26:
The part where you plug in a Trezor, enter PIN, and go straight to "select number of words", is bad UX. At minimum, there should be the recovery homescreen asking you to click a button to go to the "select number of words" screen.
(that would also solve this issue, though the only "true" solution is a Cancel button on the select number screen)

obrusvit, 2024-08-16:
IMHO, not entering "recovery mode" before the number of words is chosen is the best solution.

obrusvit, 2024-08-18:
After looking into this a little bit, the 2 suggested solutions:

1. > add a Cancel button to the "select number of words" screen
    - this would mean adding some way to close the window for TT and TS5
[image]   [image]
    - TS3 already has this choice separated into two screens and the first one has option to cancel

2. > do not go to recovery mode before number is selected 
    - this would require either setting up a parameter for default homescreen, something like this:
    ```python
    async def recovery_device(msg: RecoveryDevice) -> Success:
        # ... 
        word_count = await request_word_count(recovery_type)   
        workflow.set_default_with_args(recovery_homescreen, word_count)
    ```
    - or setting it into storage for later retrieval, something like this:
    ```python
    async def recovery_device(msg: RecoveryDevice) -> Success:
        # ... 
        word_count = await request_word_count(recovery_type)
        storage_recovery.set_word_count(word_count)
    ```

Currently, it's bad UX because 
- you are greeted with this screen after turning on the device
- even if you select the number of words and then you turn off and on, you have to select the number of words again

WDYT, @Hannsek @matejcik

Hannsek, 2024-08-19:
Cannot we "put the checkpoint" to the first screen rather than this screen with the numbers?

[image]

matejcik, 2024-08-19:
> Cannot we "put the checkpoint" to the first screen rather than this screen with the numbers?

how about dry-run recovery?

Hannsek, 2024-08-19:
We put an additional screen before the "number of words" screen in dry-run recovery.

Hannsek, 2024-08-26:
Related [issue](https://satoshilabs.slack.com/archives/CL1D61PQF/p1724532430625009).

obrusvit, 2025-01-17:
This issue has two parts:
1. The UI in the "Select number of words" screen does not offer "Cancel".
    - It makes the cancellation unintuitive (the user has to choose the number first and then cancel).
    - It's more prone to get to point 2 below because the user then tries to "Cancel" from Suite leaving them stuck.
    - Designs are needed @Hannsek @lapohoda 
2. There seems to be a bug in "RecoveryHomescreen" cancellation. If you cancel the recovery flow from host (e.g. by `C-c` when using `trezorctl`), the `UnexpectedMessageError` is handled leaving the device in recovery flow, so far so good. Now, if you try to cancel from the menu (in case of T3T1) after selecting the number of words, it does not work until you re-plug the device. 

Steps to reproduce point 2 on T3T1 emu:
- `./emu.py -e`
- host: `trezorctl device recover`
- trezor: go to number of words selection
- host: `Ctrl + c`
- trezor: select any number of words and try to "Cancel"
    - leaves the device stuck in Recovery
- restart emulator with `./emu.py`
    - now if you select any number of words and go to "Cancel", it correctly cancels the flow.

Hannsek, 2025-01-23:
The UI for this issue.

- [TS5](https://www.figma.com/design/n43caWJFGJlT167AQcxruu/branch/dMBO1tg7TJXLGe9JCWFBrH/%E3%80%94-Prod-%E3%80%95Safe-5?node-id=3075-4010&t=wJz2RFRiEOU8vAAZ-11)
- [TS3](https://www.figma.com/design/ZEoQDp4EtHaAJHvYNxExHb/branch/uwIeSr5Sscjnua6LilnyLC/%E3%80%94-Prod-%E3%80%95Safe-3?node-id=4407-2535&t=9HDf8K6yApInfDo0-11)
- [TT](https://www.figma.com/design/WbzQVEk1WzP0n9lRMJUB9S/branch/sXh0L62cKo6PAH1uXkXsyq/%E3%80%94-Prod-%E3%80%95Model-T?node-id=2348-581&t=s36hPBGqlt28fRWN-11)

ibz, 2025-02-17:
I am wondering whether we also want the cancel button in the case of "unlock repeated backup" (when the choices are just `20` and `33` words). I would say yes, but maybe there are other opinions? @matejcik @Hannsek

Hannsek, 2025-02-17:
Yes, we want this.

closed 2025-02-25 by ibz