trezor/trezor-firmware #3513: Signing PSBT with mixed input (Legacy, p2sh-Segwit, Segwit) fails on Trezor-T fw v2.6.4

brianddk, 2024-02-05:
**Describe the bug**

Using one Trezor-T initialized with SLIP-14 and a named passphrase, I created three PSBTs.  One from a P2PKH input, one from a P2SH-Segwit input and one from a P2WPKH input.  I can sign all three of these in turn, but when I combine all three inputs into one PSBT, the signing operation fails.

**Firmware version and revision**

Trezor-T Firmware v2.6.4

**Desktop/smartphone setup (please complete the following information):**

 - Wallet software: Electrum  (standalone) v4.5.2
 - OS: Windows 10 22H2

**To Reproduce**

1. Intialize the Trezor-T hardware wallet as instructed in `mk-psbt.sh` (attached)
2. Create three `testnet` Electrum wallet files using Trezor one for P2PKH, P2SH-Segwit, P2WPKH.
3. Load each wallet file ~ 1M sat each
4. Use `payto` in Electrum to make three PSBTs each with one input and the same output (see `mk-psbt.sh`)
5. Open the P2WPKH wallet file in the Desktop GUI ensuring good connectivity to Trezor
6. From the GUI load the unsigned `segwit.psbt` file created earlier
7. Use the combine operation to join/combine inputs and outputs from the other two PSBTs
8. Attempt to sign the combined PSBT and note the failure.

**Expected behavior**

For the signing operation on the mixed input PSBT to be as successful as the single input PSBTs

**Stacktrace from Electrum log**

```
20240205T002142.104544Z |     INFO | plugin.DeviceMgr | getting client for keystore
20240205T002142.104544Z |     INFO | plugin.DeviceMgr | end client for keystore
20240205T002142.246621Z |     INFO | plugin.DeviceMgr | getting client for keystore
20240205T002142.247621Z |     INFO | plugin.DeviceMgr | end client for keystore
20240205T002142.252620Z |    ERROR | gui.qt.main_window.[slip-14-pp-tz-noln] | on_error
Traceback (most recent call last):
  File "electrum\gui\qt\util.py", line 1047, in run
  File "electrum\wallet.py", line 2550, in sign_transaction
  File "electrum\plugins\trezor\trezor.py", line 96, in sign_transaction
    self.plugin.sign_transaction(self, tx, prev_tx)
  File "electrum\plugin.py", line 406, in wrapper
  File "electrum\plugin.py", line 399, in run_in_hwd_thread
  File "concurrent\futures\_base.py", line 458, in result
  File "concurrent\futures\_base.py", line 403, in __get_result
  File "concurrent\futures\thread.py", line 58, in run
  File "electrum\plugins\trezor\trezor.py", line 304, in sign_transaction
    inputs = self.tx_inputs(tx, for_sig=True, keystore=keystore)
  File "electrum\plugins\trezor\trezor.py", line 361, in tx_inputs
    assert desc
AssertionError
```

**Additional context**

If there is a better way (`trezorctl`) method to do this, please let me know.

* Related issues: https://github.com/spesmilo/electrum/issues/8868
* Attachment: [psbt-error.tar.gz](https://github.com/spesmilo/electrum/files/14160039/psbt-error.tar.gz)

prusnak, 2024-02-05:
I think you need to turn off the "safety checks" on the device. Can you please confirm @matejcik?

brianddk, 2024-02-05:
> I think you need to set the "safety checks" on the device to "prompt" via `trezorctl set safety-checks prompt`.

Tried using the retail firmware v2.6.2 and a bit backdated `trezorctl` version v0.13.5. 

Got the same stacktrace after `trezorctl set safety-checks prompt`

prusnak, 2024-02-05:
Any ideas @matejcik?

matejcik, 2024-02-05:
this is an instance of https://github.com/spesmilo/electrum/issues/8325 so closing in favor of that.

you would need to get and attach ownership proofs for your inputs via the `GetOwnershipProof` call. sadly, this is currently not even exposed in trezorctl which makes it problematic to do this by hand.

closed 2024-02-05 by matejcik

brianddk, 2024-02-05:
> sadly, this is currently not even exposed in `trezorctl` which makes it problematic to do this by hand.

Is there a work-item for this in a future release of `trezorctl`, or some other API that would allow this type of TXN to be signed?  Or should I just follow this feature in https://github.com/spesmilo/electrum/issues/8325 for the time being?

brianddk, 2024-02-06:
> some other API that would allow this type of TXN to be signed? 

@matejcik I was able to get it to work using the `sign_tx.py` snippet.  Below is the diff-deck to make it work for me.  The Trezor-T simply gave a warning

> Spending from multiple accounts.

> Continue anyway?

Which I do.  I'm assuming (perhaps naively) that this will work on mainnet as well as testnet.

```
28a29,32
> from trezorlib.ui import ClickUI
> from os import environ
> environ['PASSPHRASE'] = "slip-14-ps-lightning"
> 
55c59
<     CLIENT = get_default_client()
---
>     CLIENT = get_default_client(ui = ClickUI(passphrase_on_host = True))
70,71c74,84
<         address_n=parse_path("44h/1h/0h/0/0"),  # mvbu1Gdy8SUjTenqerxUaZyYjmveZvt33q
<         amount=1_827_955,
---
>         address_n=parse_path("49h/1h/0h/0/0"),  # 2Mz5B83aq1DurZnwCejSaZmJ1o3HHvSgPSi
>         amount= 1_000_000,
>         prev_hash=bytes.fromhex(
>             "190e47b90c9d5f66edfb60e52d1c790486cdf64d22b7326f77494734d51cf838"
>         ),
>         script_type=messages.InputScriptType.SPENDP2SHWITNESS,
>         prev_index=1,
>     ),
>     messages.TxInputType(
>         address_n=parse_path("44h/1h/0h/0/0"),  # myCdyq4ptyeYNpKz5J2kaFQEddRe11LeWw,
>         amount= 1_000_000,
73c86
<             "58d56a5d1325cf83543ee4c87fd73a784e4ba1499ced574be359fa2bdcb9ac8e"
---
>             "b6eaaaa3ee18e0515ff5032c2f186b77945461ccc7318cc7aaed96cbb80c1770"
74a88
>         script_type=messages.InputScriptType.SPENDADDRESS,
76a91,99
>     messages.TxInputType(
>         address_n=parse_path("84h/1h/0h/1/28"),  # tb1qwqkfzkhp7ezyj3re96yqv3atrll8mhuuljhtq7
>         amount= 999_635,
>         prev_hash=bytes.fromhex(
>             "ecd62ec98beb51eb2904724c5228e9790b1bcd661c0c1ce84810172cb76d15f9"
>         ),
>         script_type=messages.InputScriptType.SPENDWITNESS,
>         prev_index=0,
>     ),
78d100
< count = 255
82c104
<         amount=(1_827_955 - 10_000) // count,
---
>         amount=1_000_000 + 1_000_000 + 999_635 - 351*25,
84,85c106
<     )
<     for _ in range(count)
---
>     ),
```

matejcik, 2024-02-06:
ahh, so what you wanted is _just_ to spend from multiple account types in the same transaction? that is much easier.

You can just use the `tools/build_tx.py` script exactly as described in the examples, and fill out the correct UTXOs and matching derivation paths. Nothing more to it.

What I assumed is going on is messing with multiple independent signers, the way you described the setup as three independent Electrum wallets.

brianddk, 2024-02-06:
> What I assumed is going on is messing with multiple independent signers, the way you described the setup as three independent Electrum wallets.

Apologies on my wording, I Electrum is set up so that you can't have a Segwit and Legacy derivation in the same wallet.  That was the only reason I split them into three.  The wallet seed / passphrase / fingerprint is the same for all three.

PS... I asked @SomberNight if this correction might change the evaluation of the Electrum issue mentioned in the opening text.