trezor/trezor-firmware #3524: CI actions with UI reports fail for 3rd party contributors

matejcik, 2024-02-07:
e.g. this: https://github.com/trezor/trezor-firmware/actions/runs/7815616038/job/21320363060

the error caused by tokens / aws rights / whatever must not fail the pipeline. it should come up green, otherwise we don't know if the contribution actually failed any of the tests

assigned to @vdovhanych but cc @mmilata too if you have a solution in mind?

mmilata, 2024-03-09:
In 5064d2491a584a5acc54176f339b06c65fdaee5e, the errors are ignored, so the jobs don't fail needlessly . Of course, it would be nicer to actually see the UI changes from 3rd party contributors.

vdovhanych, 2024-03-09:
I don't think this is doable. We would basically allow deploying to our domain from the 3rd party pull requests. Even though we still need to approve the CI to run, there is a risk of overlooking something malicious.

mmilata, 2024-03-09:
OK, closing the issue then. Please reopen if my commit did not actually solve the problem.

closed 2024-03-09 by mmilata