trezor/trezor-firmware #3536: Information about account missing when sending ETH.

obrusvit, 2024-02-15:
A clear and concise description of what the bug is.
Sending flow of ETH and potentially other altcoins does not show the information about the _account_ as is shown in bitcoin sending flow. 

Sending BTC show info about account:
[image]
Similar information is not found when sending e.g. ETH. 

The problem was discovered while working on #3517, where similar screens were a part of the design: https://www.figma.com/file/PNqL15hf98oIFjLc0WCrvz/Sandbox-%E2%80%A2-Model-T?type=design&node-id=47-1280&mode=design&t=ApnCXlRR331Frs2h-4

ibz, 2024-09-12:
No. I'll have a look at it now.

bosomt, 2024-10-10:
**QA OK**

**Info:**
 - Suite version: desktop 24.11.0 (850d7ee26ec9c8274058f60bbaba0cc4c29de357)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuiteDev/24.11.0 Chrome/128.0.6613.162 Electron/32.1.2 Safari/537.36
 - OS: MacIntel
 - Screen: 1512x982
 - Device: Trezor T3T1 2.8.4 regular (revision d71d9e9c343ad04a7f28d6b02f8d3f8945af30d6)
 - Transport: BridgeTransport 2.0.33

ibz, 2025-01-28:
Re-opened, as we have a PR for model T...

PR #4175 (2024-09-24): feat(core): show account info in ETH send/stake flow
Fixes  #3536
closed 2025-02-11 by ibz