trezor/trezor-firmware #3539: Send descriptor in GetPublicKey response for Taproot accounts

mmilata, 2024-02-20:
Currently Suite has to compute the descriptor checksum, and because the checksum is computed from pubkey representation and one pubkey has multiple legal representations this can lead to the undesirable case where Suite displays different checksum than firmware. Two possibilities:

1. Send the descriptor including checksum in `PublicKey.xpub` for Taproot accounts. Disadvantage: breaks compatibility with older firmware.
2. Add new field `PublicKey.descriptor` and put the descriptor there. Advantage is that it could work for non-Taproot accounts too. Disadvantage is that Suite needs to compute the checksum itself for older firmware versions.

Previous descriptor work: #1362 #1363 #1888 #3505.

prusnak, 2024-02-20:
> 2\. Disadvantage is that Suite needs to compute the checksum itself for older firmware versions.

FWIW I am fine with Suite only showing the xpub when descriptor is not provided.

Hannsek, 2024-02-21:
Yes, I also agree that the incompatibility with old fw is not a problem here.

PR #3545 (2024-02-27): Send BIP-380 descriptor in GetPublicKey response
Closes #3539.

Turns out whoever implemented #3505 has used fingerprint of the wrong node (leaf instead of root).

Also, are we using `h` or `'` for hardened nodes? Went with `h` for now.
closed 2024-02-27 by mmilata