trezor/trezor-firmware #3575: GC9307 Screen display color and position errors.

Heclgg, 2024-03-02:
**Describe the bug**
When I change ST7789V ic to GC9307, the color is wrong, and the position is not always right.
The color is from RGB to BRG, so color is so weird.
Rotating the screen in the suite causes the position to shift. In the beginning, the screen is offset by 80 pixels upwards  towards north, and after rotation, the north- and west-facing faces appear normal, but there will be an 80-pixel offset towards south and east.

**Firmware version and revision**
I compiled it myself using the code from the repository and the model number is T.

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trezor Suite
 - OS: Windows
 - Browser: Chrome
 - Version [e.g. 90.0.2]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Settings'
2. Click on 'Device'
3. Scroll down to 'Customization'
4. Change 'Display rotation'
5. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]
[image]

**Additional context**
Add any other context about the problem here.

prusnak, 2024-03-02:
Try the following change. It turns off autodetection of the display and forces using values for GC9307.

Did it help?

``` patch
diff --git a/core/embed/trezorhal/boards/trezor_t.h b/core/embed/trezorhal/boards/trezor_t.h
index 8d6c8ba2d..2a9708c5b 100644
--- a/core/embed/trezorhal/boards/trezor_t.h
+++ b/core/embed/trezorhal/boards/trezor_t.h
@@ -16,7 +16,7 @@
 
 #include "displays/panels/lx154a2422.h"
 #include "displays/st7789v.h"
-#define DISPLAY_IDENTIFY 1
+#define DISPLAY_IDENTIFY 0
 #define DISPLAY_TE_PORT GPIOD
 #define DISPLAY_TE_PIN GPIO_PIN_12
 #define TRANSFORM_TOUCH_COORDS lx154a2422_transform_touch_coords
diff --git a/core/embed/trezorhal/stm32f4/displays/st7789v.c b/core/embed/trezorhal/stm32f4/displays/st7789v.c
index 801a08127..cb1eebac1 100644
--- a/core/embed/trezorhal/stm32f4/displays/st7789v.c
+++ b/core/embed/trezorhal/stm32f4/displays/st7789v.c
@@ -111,7 +111,7 @@ static uint32_t display_identify(void) {
   return id;
 }
 #else
-static uint32_t display_identify(void) { return DISPLAY_ID_ST7789V; }
+static uint32_t display_identify(void) { return DISPLAY_ID_GC9307; }
 #endif
 
 bool display_is_inverted() {
```

Heclgg, 2024-03-03:
No, is does not work.

Heclgg, 2024-03-03:
[image]
When downloading the firmware, it shows that. But touch position is normal. After the firmware is burned, rotate the screen direction in the suite and display it again facing north, and it will be normal.
[image]
[image]
[image]
[image]
And color is weird. 
Please.

Heclgg, 2024-03-03:
After only burning the boardloader, an Internal Error will be reported, and the screen should be red, but the screen display is blue, as shown in the figure.
[image]

---

Once I've made adjustments in this place, the screen is now red. However, the color is still abnormal after entering the firmware.

[image]
[image]
[image]

prusnak, 2024-03-03:
> No, is does not work.

Are you sure you set `DISPLAY_IDENTIFY` to `0`? Did you call `make clean` between the builds?

If you answered yes to both then try returning `DISPLAY_ID_ST7789V` from `display_identify` instead.

Heclgg, 2024-03-03:
Yes, I tried that the situation did not change.
---
[image]

Heclgg, 2024-03-04:
Is it possible that this is a problem with the big endian and little endian ?

prusnak, 2024-03-04:
> Is it possible that this is a problem with the big endian or little endian?

No. There is a lot of code which sets up different values for different drivers. You might have a display driver which is actually a hybrid of those or something similar. I see two options:
- you go through the display code and try to understand what it does, try to tweak some values to match your display
- you get another display which has ST7789 driver and just works

Heclgg, 2024-03-05:
Where is the output of 'printf()' in the main function? Is it a serial port?

TychoVrahe, 2024-03-05:
Actually, the big endian/little endian is part of the problem. 

A while ago, we started to use DMA2D graphic accelerator to do some more complex graphics and this required that we switch the display to expect little-endian data. 

However, the GC9307 does not support this, and since we did not use it in production of model Ts we decided to partially drop the support, while leaving the code in so in case someone is using it sees at least something.

If you still want to usit this driver you need to swtich the data format back to big endian (in https://github.com/trezor/trezor-firmware/blob/main/core/embed/trezorhal/stm32f4/displays/st7789v.h macro `PIXELDATA`), you should also disable usage of DMA2D, but some of the graphics, specifically model T homescreen, is not implemented without it.

The display shift is likely a different problem, probably a bug in `tf15411a_rotate` function in https://github.com/trezor/trezor-firmware/blob/main/core/embed/trezorhal/stm32f4/displays/panels/tf15411a.c

Heclgg, 2024-03-05:
Thank you so much for your patience, this is very helpful to me.
Finally, could you tell me how to use the serial port to print information? And where is the output of 'printf()' in the main (core/embed/firmware/main.c) function?

closed 2024-03-08 by Heclgg