trezor/trezor-firmware #3627: Account under "i" in receive flow is always 'ETH #1', regardless of derivation path

Hannsek, 2024-03-20:
**Describe the bug**
Account is always 'ETH #1', regardless of derivation path

**Firmware version and revision**
`2.7.0`

[image]
[image]

ibz, 2024-08-16:
The address format is `m/44'/coin_type'/account'/change/address_index`, so in both screenshots above the account is `/0` which is `#1`.

If I replace the `\0` with `\2`, I get `ETH #3`, for example...

[image]

prusnak, 2024-08-16:
> The address format is `m/44'/coin_type'/account'/change/address_index`, so in both screenshots above the account is `/0` which is `#1`.

Please sync with Suite on this. I think they use address index as accounts (like other ETH wallets do).

Hannsek, 2024-08-17:
In Trezor Suite, we use this derivation path for Ethereum when creating new accounts (addresses):`m/44'/60'/0'/0/{account_index}`

So it is like @prusnak says.

We also support discovering Ledger paths:
Ledger (live): m/44'/60'/{account_index}'/0/0 
Ledger (legacy): m/44'/60'/0'/{account_index}

closed 2024-09-16 by ibz

bosomt, 2024-10-11:
**QA OK**

**Info:**
 - Suite version: web 24.11.0 (65a8499db01e2226de44c748e725e0bf976b3ee5)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/129.0.0.0 Safari/537.36
 - OS: MacIntel
 - Screen: 1512x982
 - Device: Trezor T3T1 2.8.4 regular (revision d71d9e9c343ad04a7f28d6b02f8d3f8945af30d6)
 - Device: Trezor T2T1 2.8.4 regular (revision d71d9e9c343ad04a7f28d6b02f8d3f8945af30d6)
 - Transport: WebUsbTransport

PR #4117 (2024-09-16): fix(core/ethereum): use BIP44 variant to detect ETH account #
Fixes #3627