trezor/trezor-firmware #3641: Update micropython to support gcc 13 and Python 3.11

winksaville, 2024-03-22:
**Describe the bug**
My development environment is an Arch Linux Rolling release, hence my tools are generally the "latest" released versions and use gcc 13 and Python 3.11. Because [`vendor/micropython` v1.91.1](https://docs.micropython.org/en/v1.19.1/) is nearly 2 years old it is not compatible with gcc 13 and python 3.11.

Trying to `make build_unix` produces the following error:
```
vendor/micropython/py/stackctrl.c
vendor/micropython/py/stackctrl.c: In function ‘mp_stack_ctrl_init’:
vendor/micropython/py/stackctrl.c:32:32: error: storing the address of local variable ‘stack_dummy’ in ‘mp_state_ctx.thread.stack_top’ [-Werror=dangling-pointer=]
   32 |     MP_STATE_THREAD(stack_top) = (char *)&stack_dummy;
vendor/micropython/py/stackctrl.c:31:18: note: ‘stack_dummy’ declared here
   31 |     volatile int stack_dummy;
      |                  ^~~~~~~~~~~
In file included from vendor/micropython/py/runtime.h:29,
                 from vendor/micropython/py/stackctrl.c:27:
vendor/micropython/py/mpstate.h:323:23: note: ‘mp_state_ctx’ declared here
  323 | extern mp_state_ctx_t mp_state_ctx;
      |                       ^~~~~~~~~~~~
cc1: all warnings being treated as errors
scons: *** [build/unix/vendor/micropython/py/stackctrl.o] Error 1
make: *** [Makefile:266: build_unix] Error 2
```

**Firmware version and revision**
I'm attempting to use `make build_unix` on main:
```
(trezor-firmware-py3.11) wink@3900x 24-03-22T22:10:07.500Z:~/prgs/forks/trezor-firmware (main)
$ git log -1 --oneline
58c480f1a (HEAD -> main, origin/main, origin/HEAD) feat: Remove pyblake2
(trezor-firmware-py3.11) wink@3900x 24-03-22T22:10:17.258Z:~/prgs/forks/trezor-firmware (main)
```

**Desktop/smartphone setup (please complete the following information):**
```
wink@3900x 24-03-22T22:04:22.526Z:~
$ inxi
CPU: 12-core AMD Ryzen 9 3900X (-MT MCP-) speed/min/max: 2266/2200/4672 MHz
Kernel: 6.8.1-arch1-1 x86_64 Up: 1d 1h 44m Mem: 6.57/31.27 GiB (21.0%)
Storage: 1.82 TiB (35.3% used) Procs: 502 Shell: Bash inxi: 3.3.33
wink@3900x 24-03-22T22:04:32.923Z:~
$ gcc --version
gcc (GCC) 13.2.1 20230801
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

wink@3900x 24-03-22T22:04:37.663Z:~
$ python --version
Python 3.11.8
wink@3900x 24-03-22T22:04:43.194Z:~
$ 
```

**To Reproduce**
Steps to reproduce the behavior:
 * git clone --recurse-submodules git@github.com:trezor/trezor-firmware
 * cd trezor-firmware
 * cp ../trezor-firmware-rust-toolchain.toml rust-toolchain.toml
 * cat rust-toolchain.toml 
 * poetry install --no-root
 * cd core
 * poetry shell
 * rustc --version
 * gcc --version
 * python --version
 * make JOBS=1 build_unix
```
 wink@3900x 24-03-22T23:20:15.455Z:~/prgs/forks
$ git clone --recurse-submodules git@github.com:trezor/trezor-firmware
Cloning into 'trezor-firmware'...
remote: Enumerating objects: 173899, done.
[... 78 lines cut ...]
vendor/micropython/py/stackctrl.c: In function ‘mp_stack_ctrl_init’:
vendor/micropython/py/stackctrl.c:32:32: error: storing the address of local variable ‘stack_dummy’ in ‘mp_state_ctx.thread.stack_top’ [-Werror=dangling-pointer=]
   32 |     MP_STATE_THREAD(stack_top) = (char *)&stack_dummy;
vendor/micropython/py/stackctrl.c:31:18: note: ‘stack_dummy’ declared here
   31 |     volatile int stack_dummy;
      |                  ^~~~~~~~~~~
In file included from vendor/micropython/py/runtime.h:29,
                 from vendor/micropython/py/stackctrl.c:27:
vendor/micropython/py/mpstate.h:323:23: note: ‘mp_state_ctx’ declared here
  323 | extern mp_state_ctx_t mp_state_ctx;
      |                       ^~~~~~~~~~~~
cc1: all warnings being treated as errors
scons: *** [build/unix/vendor/micropython/py/stackctrl.o] Error 1
make: *** [Makefile:266: build_unix] Error 2
(trezor-firmware-py3.11) wink@3900x 24-03-22T23:25:00.405Z:~/prgs/forks/trezor-firmware/core (main)
$ 
```
**Expected behavior**
I expect the compile to complete without error using the latest tools.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

winksaville, 2024-03-22:
I'd like to suggest the solution is to upgrade `micropython` from v1.19.1 to [v1.22.2](https://github.com/micropython/micropython/releases/tag/v1.22.2).

prusnak, 2024-03-23:
While I am not saying we should not bump the micropython version, the correct way how to build Trezor firmware in a reproducible way is to use Docker via `./build-docker.sh` script.

prusnak, 2024-03-23:
Closing as duplicate of https://github.com/trezor/trezor-firmware/issues/2990

closed 2024-03-23 by prusnak

CashewTTS, 2024-05-12:
In the meantime, I'm working around the issues by just adding this line to core/Makefile:
```CFLAGS += -Wno-dangling-pointer -Wno-error=enum-int-mismatch```

mullroy, 2024-05-18:
> In the meantime, I'm working around the issues by just adding this line to core/Makefile: `CFLAGS += -Wno-dangling-pointer -Wno-error=enum-int-mismatch`

It appears that the CFLAGS are used by GCC, but not when Python is parsing the project files, still causing a build failure. 
My system, using the package versions bundled with the distribution:
&nbsp;&nbsp;Ubuntu 24.04 LTS
&nbsp;&nbsp;GCC & arm-none-eabi-gcc version 13.2.0
&nbsp;&nbsp;Python 3.12.3
Installed rustup separately:
apt-get install rustup
rustup install nightly-2023-04-16
&nbsp;&nbsp;Rust 1.71.0-nightly
 
I've managed to build_firmware & build_unix by embedding the override directly in the source file, as follow:

vendor/micropython/mpy-cross/main.c:
Line 333:Change int to mp_import_stat_t
&nbsp;&nbsp;&nbsp;mp_import_stat_t mp_import_stat(const char *path) {
&nbsp;&nbsp;&nbsp;(void)path;
&nbsp;&nbsp;&nbsp;return MP_IMPORT_STAT_NO_EXIST;
}

vendor/micropython/py/stackctrl.c:
void mp_stack_ctrl_init(void) {
&nbsp;&nbsp;&nbsp;volatile int stack_dummy;
&nbsp;&nbsp;&nbsp;#pragma GCC diagnostic push
&nbsp;&nbsp;&nbsp;#pragma GCC diagnostic ignored "-Wdangling-pointer"
&nbsp;&nbsp;&nbsp;MP_STATE_THREAD(stack_top) = (char *)&stack_dummy;
&nbsp;&nbsp;&nbsp;#pragma GCC diagnostic pop
}

Rust::Bindgen also needed to be updated:
core/embed/rust/Cargo.toml:
[build-dependencies.bindgen]
version = "0.69.4"
Inside the core/embed/rust directory, run: 'cargo update'