trezor/trezor-firmware #3648: flash size comparison CI job for T2B1 not working

TychoVrahe, 2024-03-26:
Seems to be comparing TT binary/elf from main with TS3/T2B1 build from PR. 

https://github.com/trezor/trezor-firmware/actions/runs/8422330162/job/23061992809?pr=3647

Hannsek, 2024-05-09:
Is this still valid?

mmilata, 2024-05-31:
The check is also unreliable for T2T1 as it compares the size with last GitLab build that passes all checks (which sometimes doesn't happen for weeks due to e.g. broken arm builders).

Instead we should push the `main` firmware images from GitHub Actions to data.trezor.io as GitLab CI will become deprecated for us.

https://satoshilabs.slack.com/archives/CL1D61PQF/p1712662716851869
https://satoshilabs.slack.com/archives/CL1D61PQF/p1717161777714039