trezor/trezor-firmware #3666: Receive flow

Hannsek, 2024-04-01:
https://www.figma.com/file/n43caWJFGJlT167AQcxruu/%E3%80%94-Prod-%E3%80%95Safe-5?type=design&node-id=26%3A4176&mode=design&t=vdrfMMQHI03AL7c0-1

mmilata, 2024-06-11:
Seems to be complete except the part that stays on after the final protobuf message, renaming the ticket to track that. Needs #3686.

matejcik, 2025-04-02:
which flows should be covered here?
all of them?

is the design with "continue in the app" up-to-date for this feature?

Hannsek, 2025-04-02:
yes, all flows with success screen.

The designs are not changing from current designs in production.

romanz, 2025-05-20:
This issue also covers transaction signature flows.

romanz, 2025-05-28:
Re-opening since we want to keep the confirmation layout open for a few seconds (#5097), even if the client sends a message immediately after the response (currently, it closes the confirmation screen).

closed 2025-06-20 by romanz