trezor/trezor-firmware #3687: Poetry install: broken dependencies

tomesink, 2024-04-07:
**Describe the bug**
After cloning the main branch poetry won't install pyyaml depenency  with python version `3.12.2` and older (see bellow).

**Firmware version and revision**
`main` branch. Just cloned  `git clone --recurse-submodules https://github.com/trezor/trezor-firmware.git`
poetry install

**Desktop setup:**
 - OS: `macOS Sonoma 14.4.1 (23E224)`
 - python: `3.12.2`  also occurs on `3.11.9`, `3.8.1` and `3.8.2`

**To Reproduce**
Steps to reproduce the behavior just follow the build instructions from docs:
1. `git clone --recurse-submodules https://github.com/trezor/trezor-firmware.git`
2. `cd trezor-firmware`
3. `poetry install`

**Expected behavior**
No error. I can proceed to
4. `cd core`

**Screenshots**
```sh
 Installing pyyaml (6.0): Failed

  ChefBuildError

  Backend subprocess exited when trying to invoke get_requires_for_build_wheel

  running egg_info
  writing lib/PyYAML.egg-info/PKG-INFO
  writing dependency_links to lib/PyYAML.egg-info/dependency_links.txt
  writing top-level names to lib/PyYAML.egg-info/top_level.txt
  Traceback (most recent call last):
    File "/Users/martin/Library/Application Support/pipx/venvs/poetry/lib/python3.12/site-packages/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
      main()
    File "/Users/martin/Library/Application Support/pipx/venvs/poetry/lib/python3.12/site-packages/pyproject_hooks/_in_process/_in_process.py", line 335, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/Users/martin/Library/Application Support/pipx/venvs/poetry/lib/python3.12/site-packages/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
      return hook(config_settings)
             ^^^^^^^^^^^^^^^^^^^^^
    File "/private/var/folders/zh/s5jmjfgn78vcz6ryn05ghls00000gn/T/tmpwonam_m1/.venv/lib/python3.12/site-packages/setuptools/build_meta.py", line 325, in get_requires_for_build_wheel
      return self._get_build_requires(config_settings, requirements=['wheel'])
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/private/var/folders/zh/s5jmjfgn78vcz6ryn05ghls00000gn/T/tmpwonam_m1/.venv/lib/python3.12/site-packages/setuptools/build_meta.py", line 295, in _get_build_requires
      self.run_setup()
    File "/private/var/folders/zh/s5jmjfgn78vcz6ryn05ghls00000gn/T/tmpwonam_m1/.venv/lib/python3.12/site-packages/setuptools/build_meta.py", line 311, in run_setup
      exec(code, locals())
    File "<string>", line 288, in <module>
    File "/private/var/folders/zh/s5jmjfgn78vcz6ryn05ghls00000gn/T/tmpwonam_m1/.venv/lib/python3.12/site-packages/setuptools/__init__.py", line 104, in setup
      return distutils.core.setup(**attrs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/private/var/folders/zh/s5jmjfgn78vcz6ryn05ghls00000gn/T/tmpwonam_m1/.venv/lib/python3.12/site-packages/setuptools/_distutils/core.py", line 185, in setup
      return run_commands(dist)
             ^^^^^^^^^^^^^^^^^^
    File "/private/var/folders/zh/s5jmjfgn78vcz6ryn05ghls00000gn/T/tmpwonam_m1/.venv/lib/python3.12/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
      dist.run_commands()
    File "/private/var/folders/zh/s5jmjfgn78vcz6ryn05ghls00000gn/T/tmpwonam_m1/.venv/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
      self.run_command(cmd)
    File "/private/var/folders/zh/s5jmjfgn78vcz6ryn05ghls00000gn/T/tmpwonam_m1/.venv/lib/python3.12/site-packages/setuptools/dist.py", line 967, in run_command
      super().run_command(command)
    File "/private/var/folders/zh/s5jmjfgn78vcz6ryn05ghls00000gn/T/tmpwonam_m1/.venv/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
      cmd_obj.run()
    File "/private/var/folders/zh/s5jmjfgn78vcz6ryn05ghls00000gn/T/tmpwonam_m1/.venv/lib/python3.12/site-packages/setuptools/command/egg_info.py", line 321, in run
      self.find_sources()
    File "/private/var/folders/zh/s5jmjfgn78vcz6ryn05ghls00000gn/T/tmpwonam_m1/.venv/lib/python3.12/site-packages/setuptools/command/egg_info.py", line 329, in find_sources
      mm.run()
    File "/private/var/folders/zh/s5jmjfgn78vcz6ryn05ghls00000gn/T/tmpwonam_m1/.venv/lib/python3.12/site-packages/setuptools/command/egg_info.py", line 550, in run
      self.add_defaults()
[... 99 lines cut ...]
So, I tried several python environments older than 3.12 (`3.11.9`, `3.8.1` and `3.8.2`). 

With these older version I got error when installing `curve25519-donna` package:

```sh
 Installing curve25519-donna (1.3): Failed

  ChefBuildError

  Backend subprocess exited when trying to invoke build_wheel

  python-src/curve25519/curve25519module.c:90:9: error: incompatible pointer to integer conversion initializing 'Py_ssize_t' (aka 'long') with an expression of type 'void *' [-Wint-conversion]
          NULL,
          ^~~~
  /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/usr/include/sys/_types/_null.h:30:15: note: expanded from macro 'NULL'
  #define NULL  __DARWIN_NULL
                ^~~~~~~~~~~~~
  /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/usr/include/sys/_types.h:52:23: note: expanded from macro '__DARWIN_NULL'
  #define __DARWIN_NULL ((void *)0)
                        ^~~~~~~~~~~
  1 error generated.
  error: command '/usr/bin/clang' failed with exit code 1
```

In sum, it is very nice dependency hell and I wonder if there is anyone who could confirm this bug and/or tell me what is the working python version setup on mac OS for the current `master` branch.

matejcik, 2024-04-16:
> well me what is the working python version setup on mac OS for the current `master` branch.

Going with nixos is the safe bet. You can [install nix](https://nixos.org/download/) on your Mac and then enter `nix-shell` in the `trezor-firmware` checkout, which should give you working versions of everything.

prusnak, 2024-04-16:
PyYAML issue fixed here:
ed25519 issue has no easy fix I am afraid ...

matejcik, 2024-04-16:
curve25519-donna _is_ used in crypto tests -- and in some strange manner too, i'm having trouble replacing it with any other library. but I don't think breaking the whole env just for two test cases is the right thing to do 

@onvej-sl you're probably the person who can best look at `test_curves.py:test_curve25519` and figure out how to drop the `curve25519-donna` dependency?
i'm thinking, we should be able to use `_ed25519.py` from trezorlib, but any way I call things i'm getting bad results

onvej-sl, 2024-04-16:
> i'm thinking, we should be able to use _ed25519.py from trezorlib, but any way I call things i'm getting bad results

This doesn't work because _ed225519.py implements ed25519 instead of curve25519. Nevertheless, both the curves are birationally equivalent. This means the underlying groups are isomorphic and the isomorphism is easily computable.

> @onvej-sl you're probably the person who can best look at test_curves.py:test_curve25519 and figure out how to drop the curve25519-donna dependency?

I think we can use the package [cryptography](https://cryptography.io/en/latest/) which is already used in device tests.

onvej-sl, 2024-04-16:
<details>
<summary>See the diff:</summary>

```diff
--- a/crypto/tests/test_curves.py
+++ b/crypto/tests/test_curves.py
@@ -5,9 +5,9 @@ import hashlib
 import os
 import random
 
-import curve25519
 import ecdsa
 import pytest
+from cryptography.hazmat.primitives.asymmetric.x25519 import X25519PrivateKey
 
 
 def bytes2num(s):
@@ -344,17 +344,17 @@ def test_validate_pubkey_direct(point):
 def test_curve25519(r):
     sec1 = bytes(bytearray(r.randbytes(32)))
     sec2 = bytes(bytearray(r.randbytes(32)))
-    pub1 = curve25519.Private(sec1).get_public()
-    pub2 = curve25519.Private(sec2).get_public()
+    pub1 = X25519PrivateKey.from_private_bytes(sec1).public_key()
+    pub2 = X25519PrivateKey.from_private_bytes(sec2).public_key()
 
     session1 = r.randbytes(32)
-    lib.curve25519_scalarmult(session1, sec2, pub1.public)
+    lib.curve25519_scalarmult(session1, sec2, pub1.public_bytes_raw())
     session2 = r.randbytes(32)
-    lib.curve25519_scalarmult(session2, sec1, pub2.public)
+    lib.curve25519_scalarmult(session2, sec1, pub2.public_bytes_raw())
     assert bytearray(session1) == bytearray(session2)
 
-    shared1 = curve25519.Private(sec2).get_shared_key(pub1, hashfunc=lambda x: x)
-    shared2 = curve25519.Private(sec1).get_shared_key(pub2, hashfunc=lambda x: x)
+    shared1 = X25519PrivateKey.from_private_bytes(sec2).exchange(pub1)
+    shared2 = X25519PrivateKey.from_private_bytes(sec1).exchange(pub2)
     assert shared1 == shared2
     assert bytearray(session1) == shared1
     assert bytearray(session2) == shared2
@@ -362,10 +362,10 @@ def test_curve25519(r):
 
 def test_curve25519_pubkey(r):
     sec = bytes(bytearray(r.randbytes(32)))
-    pub = curve25519.Private(sec).get_public()
+    pub = X25519PrivateKey.from_private_bytes(sec).public_key()
     res = r.randbytes(32)
     lib.curve25519_scalarmult_basepoint(res, sec)
-    assert bytearray(res) == pub.public
+    assert bytearray(res) == pub.public_bytes_raw()
 
 
 def test_curve25519_scalarmult_from_gpg(r):
```
<details>

prusnak, 2024-04-16:
> iiiiii don't think we even _need_ the `ed25519` package..........

We do not, but we might need curve-ed25519 which requires ed25519.

closed 2024-04-17 by matejcik