trezor/trezor-firmware #3690: Invalid Signature when using signer key to sign the transaction source on Stellar

shuangqiu123, 2024-04-09:
**Describe the bug**
I encountered an issue with Trezor Connect while attempting to sign a Stellar transaction using a Trezor One. The source wallet is generated from Trezor, and it contains a signer which is also generated from Trezor. The process involves utilizing the TrezorConnect.stellarSignTransaction method to initiate the transaction signing with the signer. Despite successfully invoking the signing method, the process results in an "invalid signature" error upon adding the signature to the transaction.

**Info:**

-   Browser: Chrome, 123.0.6312.87
-   Browser Installation Method: downloaded installer
-   OS: Macos 14.4.1
-   Bridge: version: 2.0.27
-   Firmware Version: Trezor one, firmware version 1.12.1
-  Trezor connect web npm version: 9.2.2
-  Trezor  connect stellar plugin version: 9.0.3
-  Stellar sdk version: 11.3.0

**How to reproduce**
Steps to reproduce the behavior:
1. Use TrezorConnect.stellarGetAddress to get an account. Use Stellar SDK with Friendbot to fund the account. This will be the source wallet.
2. Use Stellar SDK with TransactionBuilder to build a transaction with a payment Operation.
3. Use TrezorConnect.stellarGetAddress to get a different account, it will be used as the signer.
4.  Use Stellar SDK with TransactionBuilder to build a transaction with a setOption operation, set the following:
    - min, medium, max threshold to 1
    - master key weight to 0
    - the signer address with weight 1
5. Use transformTrezorTransaction from Trezor Stellar Plugin to transform the transaction by using the path of the source wallet. Use TrezorConnect.stellarSignTransaction to sign the transformed transaction. Add the signature and submit it to the testnet. This adds the signer into the source wallet.
6. Use transformTrezorTransaction to transform the payment transaction from step 2 by using the path of the signer. Use TrezorConnect.stellarSignTransaction to sign the transformed transaction. 

Here is the code:
```
   // get an account for the source wallet
    const connectWallet = async () => {
      let address = "";
      try {
        const result = await TrezorConnect.stellarGetAddress({
          path: "m/44'/148'/" + selectedNumber + '\'',
          showOnTrezor: false,
          useEmptyPassphrase: true
        });

        if (result.success) {
          address = result.payload.address;
          console.log('Stellar Address:', address);
          setXlmAddress(address);
          setIsConnected(true);
          server.accounts().accountId(address)
          .call()
          .then((response)=> {
            console.log(response);
            setWalletBalance(response.balances[0].balance);
          })
          .catch((error)=> {
            server.friendbot(address).call()
            .then((response)=> {
              console.log(response);
            })
          });
        } else {
          console.error('Error:', result.payload.error);
        }
      } catch (error) {
        console.error('Error:', error);
      }
    }
    // generate payment transaction
    const generate = async () => {
      const query = await server.accounts().accountId(xlmAddress).call();
      const senderAccount = new Account(xlmAddress, query.sequence);

      const tx = new TransactionBuilder(senderAccount, {
        fee: "100",
        networkPassphrase: "Test SDF Network ; September 2015",
[... 92 lines cut ...]
#### Signing with master key or a KeyPair Signer works
The signature is valid when signing with the master key, continued from the example above,
Valid XDR signed by GARO32RIIDX6EED6JKAMIWUK4CJGIRJVY7ALWIJ22DVI6TVX2NJYPDUS: ```AAAAAgAAAAAi7eooQO/iEH5KgMRaiuCSZEU1x8C7ITrQ6o9Ot9NThwAAAGQADmo5AAAAAgAAAAEAAAAAAAAAAAAAAABmDjAOAAAAAAAAAAEAAAABAAAAACLt6ihA7+IQfkqAxFqK4JJkRTXHwLshOtDqj06301OHAAAAAQAAAAA3RW0ndjA9zBQ/RfaN1B2DEvSi4wgrlBqsXkOgcKOSigAAAAAAAAAAAJiWgAAAAAAAAAABt9NThwAAAEApT+QxVD0mj8VnX3ZtZ5v1Bjv4D7K7cIGuE/F9bNJ9/zPPGLK8DKs1GAJwAiHOb/bL7pDHNFWI/Gad8D0t6d4B```
[image]

The signature is valid when signing with a KeyPair signer
Source Wallet: GDQA7QXCACLZWIJUIBHGBFNTOXDFDH3FHIK47BB3WU3OOWDXEXMOUEZV
Signer Public Key: GAKTLJG7O5YWPUMRTH3L42NY3655FDYXLHAN346IEMTGGCBJTQBVO5PV
Signer Secret Key: SDAZQA3WYLOQWENENXYKGJ4E6ZOR3YC5KOLJZNJQR4UMSHOZEBESHBIL
Unsigned XDR: ```AAAAAgAAAADgD8LiAJebITRATmCVs3XGUZ9lOhXPhDu1NudYdyXY6gAAAGQADmsVAAAAAQAAAAEAAAAAAAAAAAAAAABmDjR3AAAAAAAAAAEAAAABAAAAAOAPwuIAl5shNEBOYJWzdcZRn2U6Fc+EO7U251h3JdjqAAAAAQAAAAA3RW0ndjA9zBQ/RfaN1B2DEvSi4wgrlBqsXkOgcKOSigAAAAAAAAAAAJiWgAAAAAAAAAAA```
Signed XDR:
```AAAAAgAAAADgD8LiAJebITRATmCVs3XGUZ9lOhXPhDu1NudYdyXY6gAAAGQADmsVAAAAAQAAAAEAAAAAAAAAAAAAAABmDjR3AAAAAAAAAAEAAAABAAAAAOAPwuIAl5shNEBOYJWzdcZRn2U6Fc+EO7U251h3JdjqAAAAAQAAAAA3RW0ndjA9zBQ/RfaN1B2DEvSi4wgrlBqsXkOgcKOSigAAAAAAAAAAAJiWgAAAAAAAAAABKZwDVwAAAEA2XdAFNKzVyPQH9qJx8xSp1byDyYmWNRrWi2AV1OyeKlc1RO3rs6to8awIC58hsBxFZLrAI/bFwUCXtibyeZoD```
[image]

#### Reproduced on Stellar Laboratory 
The same problem can also be reproduced on the Stellar Laboratory, with the source wallet generated by random Keypair. 
1. Generate a key pair on Stellar Lab and fund the wallet by Friendbot
8. Create a transaction with SetOptions operation that sets 
    - min, medium, max threshold to 1
    - master key weight to 0
    - a signer generated from the Trezor with weight 1 (e.g. from path m/44'/148'/0')
9. Sign the transaction on the Sign Transaction tab with the master key pair secret, and submit the returned signed XDR to the testnet
10. Create another transaction with a operation (it can be any transaction), export its XDR
11. Sign the transaction on the Sign Transaction tab with the path of the Trezor Signer, and submit the returned XDR to the testnet.
12. An error tx_bad_auth will occur

mroz22, 2024-04-09:
ok, just to clarify, is there a firmware version where it returns correct results? are you able to identify the specific firmware version when it broke?

overcat, 2024-04-09:
> ok, just to clarify, is there a firmware version where it returns correct results? are you able to identify the specific firmware version when it broke?

I think all T1 firmware returns incorrect signatures, this scenario has not been covered by test cases, I have added it in the PR #3691.

PR #3691 (2024-04-10): fix(legacy): resolves the issue of incorrect signature generation when the Stellar transaction source account differs from the signing account.
fix #3690
closed 2024-04-10 by matejcik