trezor/trezor-firmware #3697: Czech translation bugs

bosomt, 2024-04-10:
**Describe the bug**
Noticed Czech translation bugs as it was included in TS app yesterday.
- unlock attempts text is overlapping count of remaining count
- verifying pin text  is not translated

**Info:**
 - Suite version: desktop 24.4.1 (979558244707a3d9d577cf9afff6a860c48b137d)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/24.4.1 Chrome/118.0.5993.159 Electron/27.3.8 Safari/537.36
 - OS: MacIntel
 - Screen: 1512x982
 - Device: Trezor T2T1 2.7.0 regular (revision 45e8a842a31e62a6d43d7f6ccac62a45e1198ef0)
 - Transport: BridgeTransport 2.0.33

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
[image]

[image]

bosomt, 2024-04-10:
one more translation missing 
[image]

STew790, 2024-04-10:
There is also "Processing" when setting up a wipe code for example.

matejcik, 2024-04-10:
> There is also "Processing" when setting up a wipe code for example.

#3610

closed 2024-05-09 by mmilata