trezor/trezor-firmware #3765: Replace T3T1 debug `caPubKey`

komret, 2024-05-02:
Debug value for `caPubKey` should be:
 `04829e8965018feb542e9236c9b2ce08f864a55ed9183d0259564f0e05345b04676a0bef36c59d21d3c24868b5601f0b1193a6bfcf6d814e1cfb79c2256a05e953`.
This is necessary to pass the check in Suite debug mode when emulator is used. Current value is:
`04ba6084cb9fba7c86d5d5a86108a91d55a27056da4eabbedde88a95e1cae8bce3620889167aaf7f2db166998f950984aa195e868f96e22803c3cd991be31d39e7`
It is the same key as for T2B1 - therefore validation fails [here](https://github.com/trezor/trezor-suite/blob/89c1cd5d8c2afe0bdac9d73cdf100ccaaa0b42f9/packages/connect/src/api/firmware/verifyAuthenticityProof.ts#L120).

Suite expects [these keys](https://github.com/trezor/trezor-suite/blob/a04421ba2342bddce4b5638a2508cb347b1387fc/packages/connect/src/data/deviceAuthenticityConfig.ts#L94).

matejcik, 2024-05-06:
Just so you know, this turns out to be more difficult than anticipated. The staging environment will refuse to sign emulator's certificate because it is not issued by Optiga. We need to grab the staging HSM, install a modified firmware that lifts this restriction, and then sign the certificate.

komret, 2024-05-06:
> Just so you know, this turns out to be more difficult than anticipated. The staging environment will refuse to sign emulator's certificate because it is not issued by Optiga. We need to grab the staging HSM, install a modified firmware that lifts this restriction, and then sign the certificate.

Ok, I thought it would be straightforward since it works for T3B1. Worst case scenario, we can skip the authenticity checks until this is implemented, it is possible to set that in Suite settings.

Also, i notice that debug rootPubKey in Suite is wrong, the corect value is `04e48b69cd7962068d3cca3bcc6b1747ef496c1e28b5529e34ad7295215ea161dbe8fb08ae0479568f9d2cb07630cb3e52f4af0692102da5873559e45e9fa72959 ` is per https://github.com/trezor/trezor-suite/issues/12001#issuecomment-2051803935, I am going to fix it now.

matejcik, 2024-05-06:
> Ok, I thought it would be straightforward since it works for T3B1. Worst case scenario, we can skip the authenticity checks until this is implemented, it is possible to set that in Suite settings.

that's because T3B1 is (currently) using the same authority root as T2B1, no? so we send the literal same certificate and it passes the checks on your side.
this might change in the future too btw, if we decide to use a different root for T3B1

komret, 2024-05-06:
Sorry, I meant T2B1.

PR #3944 (2024-06-25): Include a valid staging certificate for TS5 emulator
fixes #3765 

includes certificate verification code and supersedes #3464
@andrewkozlik please have a look, I believe all your comments from #3464 are applied here
closed 2024-06-25 by matejcik