trezor/trezor-firmware #3773: Emulator build fails under nix-shell environment on Mac OS X

tomesink, 2024-05-05:
**Describe the bug**
Following the [docs](https://docs.trezor.io/trezor-firmware/core/build/emulator.html) for Mac OS X: 

`Consider using Nix. With Nix all you need to do is nix-shell.`

After running `nix-shell` in the root directory and subsequently run `make build_unix` under `nix-shell` environment in the `core` folder the build fails with an Error.

**Firmware version and revision**
Current (latest commit is `f8be07b`) `main` branch.

**Desktop setup :**
 - OS: `macOS Sonoma 14.4.1 (23E224)`

**To Reproduce**
Steps to reproduce the behavior:
1. `git clone https://github.com/trezor/trezor-firmware.git`
2. `cd trezor-firmware`
3. run `nix-shell` command
4. `rustup default nightly &&
rustup update`
5. `cd core`
6. `make build_unix`

**Expected behavior**
Build is successful.

**Screenshots**
```sh
[nix-shell]$ make build_unix
python ./translations/order.py
make: python: No such file or directory
make: *** [Makefile:229: translations] Error 127
```

**Additional context**
Obviously make has troubles finding the python executable, which is weird because my python is working inside the nix :
```
[nix-shell]$  python --version
Python 3.12.2
```
Also running the command manually without `make` works:
```
[nix-shell]$   python ./translations/order.py
No new items found.
```

My relevant `PATH` section for NIX is as follows taking the python from homebrew (and aliasing `python3` to `python` in my `.bashrc`). I am not sure if nix shell should provide the python binary as well

`/nix/store/5qc7bwpcxnc21zgqr8kzb0r84p9g8wp5-bash-interactive-5.2p26/bin:/nix/store/n4ry38iwq6y9wy7fpal3qpw7kzq0f1l1-SDL2-2.26.4-dev/bin:/nix/store/24myb0ky5zv2zfkm6lwrijiavly13cf7-bash-5.2-p15/bin:/nix/store/zx2z363pd48rdjy6lj8b2k2ig6khlz7h-bloaty-1.1/bin:/nix/store/xc7bn2vf86idgway6247vff4fny7j2xy-check-0.15.2/bin:/nix/store/5987rn3gd3zpvn4yrnn6v5gki6q5b2lb-crowdin-cli-3.10.1/bin:/nix/store/5w24c6rwqxh3qqck8gqlf0xil9qjgaac-curl-8.0.1-dev/bin:/nix/store/dlgmqvy4ir640bn8677l00w8g9dvd0vv-brotli-1.0.9/bin:/nix/store/vyfbsm9nqq3m6dlvkw961r4xyl9swgyq-libkrb5-1.20.1-dev/bin:/nix/store/f8kydfjbs361yddiz9phvjipp52yd37a-libkrb5-1.20.1/bin:/nix/store/mqyv0amgq6lkazh2l6gxhxn8bz93rm9m-nghttp2-1.51.0-bin/bin:/nix/store/yp1sp0qbw4gv93xqsg1k4gdqanly7f8n-libidn2-2.3.4-bin/bin:/nix/store/9ywskhzaw019f5wfyip2b6ymn489m50w-openssl-3.0.8-bin/bin:/nix/store/w4jxhxdpkl45al69ypwky28crm59pizw-zstd-1.5.4-bin/bin:/nix/store/w6c4dhlmdpdh5vjr46yygngwqysjp2hx-zstd-1.5.4/bin:/nix/store/k4byv5574j908n4r4biwaxmi0lrlb2xm-curl-8.0.1-bin/bin:/nix/store/vy3rfn5qbncvzpiyhmpz0ssb47y0wwld-editorconfig-checker-2.7.0/bin:/nix/store/xgcc4n7j2prcab6w4wkmmmlrr57i7fiy-gcc-arm-embedded-12.2.rel1/bin:/nix/store/9vcslxkvzg921lrd8s3nchv1xqsskkgs-git-2.39.2/bin:/nix/store/2yqb8xip2jhmxn1nn5bq2mf6kz13qmsa-git-subrepo-0.4.5/bin:/nix/store/d1bybmym70qv03h82ij7sc7055b75zj0-gnumake-4.4.1/bin:/nix/store/ygsjmzgcmg3f0r53aq3a19shz5g4mggr-graphviz-7.1.0/bin:/nix/store/l9qq4sdbrbhq62l0pl3k8fp2b6ihwil3-libjpeg-turbo-2.1.4-bin/bin:/nix/store/a7vwg38q9bhx82x1lphskdwjg5jdqhxy-clang-wrapper-14.0.6/bin:/nix/store/whsim4mma5i4b8q6pfl86svb5n8nvx78-clang-14.0.6/bin:/nix/store/v2v8ljfaid7icj00ay6lyv6k57xirak8-coreutils-9.1/bin:/nix/store/h2nf6sn2z9qm64hcngsb9mh528gi49g9-cctools-binutils-darwin-wrapper-973.0.1/bin:/nix/store/zrqd8yv2kmv460zyrgrlb2514nq4fk9m-cctools-binutils-darwin-973.0.1/bin:/nix/store/98wlwihggxczhy9lmiplp0gja8c3y6sb-pkg-config-wrapper-0.29.2/bin:/nix/store/yc0v8y4lk5wbxzz9rnn24sgr9lk14vj2-python3.10-poetry-1.4.2/bin:/nix/store/8zn5nwgkcslyxx48dc5jnfkz8kdnddch-protobuf-3.19.6/bin:/nix/store/v96485r4r725pvqx8k0q4fkf9573hqv6-pyright-1.1.204/bin:/nix/store/acpfbnp11iqpng7y4xpixs6p3k29ambq-rust-minimal-1.70.0-nightly-2023-04-14-bin/bin:/nix/store/arz0b8n3c4l9p09d9y04j5jlap7yilnh-wget-1.21.3/bin:/nix/store/rar52nciiy0dkv5hjxmmfvm69z0sg6l1-moreutils-0.67/bin:/nix/store/79w1425pmd99cp6s43rknig2i8k3nljn-perl-5.36.0/bin:/nix/store/9sxdgcaf7mwxgz2l2bgfnriwni46a7wa-libiconv-50/bin:/nix/store/v2v8ljfaid7icj00ay6lyv6k57xirak8-coreutils-9.1/bin:/nix/store/cph6k98y8sjmr0mis2y60s56b3sdpxj1-findutils-4.9.0/bin:/nix/store/pcfw2fnaxcxxs7ln4hy1jaaqgb8rmdhj-diffutils-3.9/bin:/nix/store/n023kyx1zmzdjqhc5jwb2vrlxmgm3pzv-gnused-4.9/bin:/nix/store/ynp1z7azqz6xcs9x3isgpm3ry4526nx0-gnugrep-3.7/bin:/nix/store/q4zlavvz0x1iyx7mrv4zwkz7lk77yams-gawk-5.2.1/bin:/nix/store/swrn8a0pzdvyqb9k4snj0f81qi2zj482-gnutar-1.34/bin:/nix/store/mbxlf6szaixjhqw07k139jam6g5vb005-gzip-1.12/bin:/nix/store/28azg0h5mighcayy28x1shgaz6wbw3nr-bzip2-1.0.8-bin/bin:/nix/store/d1bybmym70qv03h82ij7sc7055b75zj0-gnumake-4.4.1/bin:/nix/store/24myb0ky5zv2zfkm6lwrijiavly13cf7-bash-5.2-p15/bin:/nix/store/aa17c5qkavymj717hif7hsdyhglb1ngd-patch-2.7.6/bin:/nix/store/yvdv2x7z7qjhqzym5zndnimisg2b74nq-xz-5.4.2-bin/bin:/nix/store/3jsn64538qqfjyix6z89cs1ippigmf8q-file-5.44/bin:/Users/martin/.nix-profile/bin:/nix/var/nix/profiles/default/bin:/Users/martin/.nix-profile/bin:/nix/var/nix/profiles/default/bin
`

I am wondering if an additional config must be done on my/user side to make things works. In this case it would be a documentation bug. 

But nix-shell is supposed to deliver fully operational environment but given the fact the Nix does not contain its own version of python, I'd suspect this might be the problem (but I am not familiar with Nix so I might be wrong).

Hannsek, 2024-05-05:
Try following this:
1. `cd trezor-firmware`
2. `git branch` (check in which branch you are)
3. `nix-shell`
4. `poetry shell`
5. `poetry install`
6. `cd core`
7. `make clean` & `make vendor`
8. `make build_unix`(`TREZOR_MODEL=R make build_unix` for model Safe 3)
9. `./emu.py -e`

tomesink, 2024-05-05:
> Try following this:
> 
> 1. `cd trezor-firmware`
> 2. `git branch` (check in which branch you are)
> 3. `nix-shell`
> 4. `poetry shell`
> 5. `poetry install`
> 6. `cd core`
> 7. `make clean` & `make vendor`
> 8. `make build_unix`(`TREZOR_MODEL=R make build_unix` for model Safe 3)
> 9. `./emu.py -e`

This works! Also solves #3774  I would never thought I have to use **both** nix and poetry shell.

I suggest use this issue as a trigger for a new documentation PR because the build process and nix+poetry shell magic should be clearly documented.

matejcik, 2024-05-06:
@tomesink care to suggest a concrete improvement? your link has this at the very top:

[image]

tomesink, 2024-05-06:
@matejcik "all you need to do is `nix-shell`" is what confused me. I simply did not realise I still need to enter poetry even with nix. 

[image]

So i'd suggest to make it crystal clear with sentence like:

`With Nix all you need to do is nix-shell. (You still need to use poetry shell as well)`

Or do not any change and in case some other has similar problem with understanding the documentation, he would find this issue eventually :)

closed 2024-05-09 by Hannsek