trezor/trezor-firmware #3821: Additional symbols on Trezor display during setup in Czech firmware

rozisats, 2024-05-14:
In Czech firmware, during the setup - Know your Trezor/Poznej svůj Trezor on a step that says "Posunutí obrazovky, Levým tlačítkem se posunete nahoru" , there are some additional symbols on the left side of the screen around "Le" part of the word "Levým". This was replicated twice, bootloader wipe did not resolve this issue.

[image]

bosomt, 2025-03-12:
Still happening

[image]

 - Device: Trezor T3B1 2.8.9 bitcoin-only (revision fad9682201cf9289bba2adb66e6e07ed1cf78936)

evgenysl, 2025-05-06:
:x: Same bug with Portuguese language:

[image]

:x: Spanish language word SCREEN - PANTALLA is moved to next line

[image]

:white_check_mark: No issues with English, German and French because there's less text on that screen.

romanz, 2025-05-06:
Reproduces on emulator:

[image]
[image]

bosomt, 2025-06-18:
**QA OK**

tested on [main debug](https://github.com/trezor/trezor-firmware/actions/runs/15720057569/job/44298983532)
`trezorctl set language translation-T3B1-cs-CZ-2.9.0-unsigned.bin`
`trezorctl set language translation-T3B1-es-ES-2.9.0-unsigned.bin`
with `trezorctl device tutorial`
T3B1     revision: 20 bytes 8fbe0051861f081809acd80033eba4a519a2bdfd

[image]

[image]