trezor/trezor-firmware #3838: Building on Aarch64 (GCC 13 and Python 3.12)

mullroy, 2024-05-17:
This report is related to bug report #3641

I've build the emulator (build_unix) & firmware (build_firmware) in /core using a native Ubuntu 22.04 system.
I'm running the code on the following system:
&nbsp;&nbsp;&nbsp;Distribution: Ubuntu 22.04
&nbsp;&nbsp;&nbsp;rustc 1.71.0-nightly  (5cdb7886a 2023-04-15)
&nbsp;&nbsp;&nbsp;rustfmt 1.5.2-nightly (5cdb7886 2023-04-15)
&nbsp;&nbsp;&nbsp;perl 5, version 34
&nbsp;&nbsp;&nbsp;SCons: v4.7.0
&nbsp;&nbsp;&nbsp;gcc Ubuntu 12.3.0
&nbsp;&nbsp;&nbsp;arm-none-eabi-gcc (Arm GNU Toolchain 12.3.Rel1)  Manually installed.
&nbsp;&nbsp;&nbsp;(Version 10.8 shipped with package manager did not build the firmware)
&nbsp;&nbsp;&nbsp;Python 3.10.12
&nbsp;&nbsp;&nbsp;clang version 14.0.0
&nbsp;&nbsp;&nbsp;Trezorlib:Build & installed ver 0.13.9 from the ./python subdir included with the project 
&nbsp;&nbsp;&nbsp;(pip3 installs ver. 0.13.8, which doesn't have the translation functions)
&nbsp;&nbsp;&nbsp;Poetry version 1.1.12

The firmware build was successfull. Build reported: Signature is DEVEL, hashes are VALID

The unix build was successfull. I can run the emulator (./emu.py -s) which brings up the Trezor T GUI window. I can successfully interact with it using Trezor Suite 24.4.3 adding launch switch : --bridge-dev 
_So far, so good_

I'd like to build the code on my **ARM64** board. I'm running 24.04LTS
I'm developing in a chroot environment with a minimal root filesystem (noble-base-arm64.tar.gz). proc,sys & dev is mounted to the native system. I've used the same version of rust & gcc cross compiler as on the host system:

&nbsp;&nbsp;&nbsp;rustc 1.71.0-nightly (5cdb7886a 2023-04-15)  
&nbsp;&nbsp;&nbsp;rustfmt 1.5.2-nightly (5cdb7886 2023-04-15)
&nbsp;&nbsp;&nbsp;perl 5, version 38
&nbsp;&nbsp;&nbsp;SCons: v4.7.0
&nbsp;&nbsp;&nbsp;gcc 13.2.0
&nbsp;&nbsp;&nbsp;arm-none-eabi-gcc (15:13.2.rel1-2) 
&nbsp;&nbsp;&nbsp;Python 3.12.3
&nbsp;&nbsp;&nbsp;clang version 18.1.3
&nbsp;&nbsp;&nbsp;trezorlib 0.13.9 (From project ./python source)
&nbsp;&nbsp;&nbsp;Poetry version 1.8.3

Building the code triggers similar errors that are covered in bug report #3641
Adding 'CFLAGS += -Wno-dangling-pointer -Wno-error=enum-int-mismatch' to the core/Makefile
still triggered these errors (weird)

Manual work around:
  SCons SyntaxWarning: invalid escape sequence '\('
    Edit core/SConscript.unix
    Line 532: ('MICROPY_TREZOR_MEMPERF', '(1)')
  The dangling pointer:
    Edit vendor/micropython/py/stackctrl.c
    Line 31: static volatile int stack_dummy;
  The enum/int error:
    Edit vendor/micropython/mpy-cross/main.c
    Line 333: mp_import_stat_t mp_import_stat(const char *path) {
    
Buidilng the firmware: (build_firmware)
Far into the build I get this error reported by micropython application mpy-cross:
  vendor/micropython/mpy-cross/mpy-cross -O1 -o build/firmware/src/trezor/strings.mpy -s trezor/strings.py build/firmware/src/trezor/strings.i
  **FATAL: uncaught NLR 0x5501b513e0**
Comparing strings.mpy between the native successfull build vs the failed build, the first parts of the output is identical. The failed output lacks the last part of the file, compared to the successfull build.

Building the unix emulator: (build_unix)
  An error occurs when RUST compiles trezor_lib v0.1.0
  The installed version of RUST is identical between my native (Intel x86) and chroot (AArch64) environments.
  The stack backtrace item #21 shows that bindgen (v0.60) triggered the ident error
  
  `Compiling trezor_lib v0.1.0 (/home/ubuntu/trezor-firmware.git/core/embed/rust)
  error: failed to run custom build command for `trezor_lib v0.1.0 (/home/ubuntu/trezor-firmware.git/core/embed/rust)`
  Caused by:
  process didn't exit successfully: `/home/ubuntu/trezor-firmware.git/core/embed/rust/../../build/unix/rust/release/build/trezor_lib-02ed1c5d7619a0fa/build-script-build` (exit status: 101)
--- stderr
  thread 'main' panicked at '"_mp_obj_fun_builtin_fixed_t_union_(unnamed_at___/__/vendor/micropython/py/obj_h_981_5)" is not a valid Ident', /home/ubuntu/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro2-1.0.29/src/fallback.rs:702:9

[... 86 lines cut ...]
                                 at /rustc/5cdb7886a5ece816864fab177f0c266ad4dd5358/library/core/src/ops/function.rs:287:13
    43:       0x550038fddc - std::panicking::try::do_call::hc3bf3f28055af3ac
                                 at /rustc/5cdb7886a5ece816864fab177f0c266ad4dd5358/library/std/src/panicking.rs:485:40
    44:       0x550038fddc - std::panicking::try::h75551ba75ca8898f
                                 at /rustc/5cdb7886a5ece816864fab177f0c266ad4dd5358/library/std/src/panicking.rs:449:19
    45:       0x550038fddc - std::panic::catch_unwind::hf01f5d0a7e54967d
                                 at /rustc/5cdb7886a5ece816864fab177f0c266ad4dd5358/library/std/src/panic.rs:140:14
    46:       0x550038fddc - std::rt::lang_start_internal::{{closure}}::h8cca7b31051c7f42
                                 at /rustc/5cdb7886a5ece816864fab177f0c266ad4dd5358/library/std/src/rt.rs:148:48
    47:       0x550038fddc - std::panicking::try::do_call::h6e56ad44fe1adbd0
                                 at /rustc/5cdb7886a5ece816864fab177f0c266ad4dd5358/library/std/src/panicking.rs:485:40
    48:       0x550038fddc - std::panicking::try::h20c9bdeb00e113e4
                                 at /rustc/5cdb7886a5ece816864fab177f0c266ad4dd5358/library/std/src/panicking.rs:449:19
    49:       0x550038fddc - std::panic::catch_unwind::hb960463dcc3f963c
                                 at /rustc/5cdb7886a5ece816864fab177f0c266ad4dd5358/library/std/src/panic.rs:140:14
    50:       0x550038fddc - std::rt::lang_start_internal::h0384baf16f174507
                                 at /rustc/5cdb7886a5ece816864fab177f0c266ad4dd5358/library/std/src/rt.rs:148:20
    51:       0x5500047428 - std::rt::lang_start::h524396be0e9b1ec2
                                 at /rustc/5cdb7886a5ece816864fab177f0c266ad4dd5358/library/std/src/rt.rs:165:17
    52:       0x5500045528 - main
    53:       0x5501dc84c4 - <unknown>
    54:       0x5501dc8598 - __libc_start_main
    55:       0x5500041f70 - _start
    56:                0x0 - <unknown>
  `

closed 2024-05-17 by matejcik

mullroy, 2024-05-17:
Hi, thank you for the feedback. 
I've made some progress with my build. In core/embed/rust/Cargo.toml, change the version for 'bindgen' from 0.60.1 to 0.69.4.
Running 'make build_unix' succeeds 

Launching the emulator (./emu.py) also crashes with the same error as just building the firmware (make build_firmware) on its own:
uncaught NLR 0x5501b513e0

Running the rust tests:
Running make test_rust fails with 2 types of errors:
1) vendor/micropython/py/qstr.h has a '#include "genhdr/qstrdefs.generated.h', but can't locate
the genhdr directory. This is temporarily solved by copying from vendor/micropython/mpy-cross/build/genhdr/qstrdefs.generated.h to vendor/micropython/py/genhdr
2) By creating the work around in #1, the rust build finishes. 
Right at the end 'ar' tries to access the library: libcore_lib.a, which doesn't exist inside the build/unix/rust/.... build output
This error is also present on the native system. Does the rust test work?

matejcik, 2024-05-17:
the "uncaught NLR" is most likely an uncaught exception from Micropython that falls deep enough that nothing prints it.
given that you messed with the `stack_dummy`, my best guess would be that the stack overflow check is now broken and Python is failing with something like a stack overflow?

> * Include path:
>   vendor/micropython/py/qstr.h has a '#include "genhdr/qstrdefs.generated.h', but can't locate
>   the genhdr directory. This is temporarily solved by copying from vendor/micropython/mpy-cross/build/genhdr/qstrdefs.generated.h to vendor/micropython/py/genhdr

that's because your genhdr is failing and not generating the file .) you copied _something_, probably not the thing you wanted

>   Right at the end 'ar' tries to access the library: libcore_lib.a, which doesn't exist inside the build/unix/rust/.... build output
>   This error is also present on the native system.

yeah, no idea about what this is about. i'm thinking the Rust build is actually failing somewhere (else) and that failure is hidden from you but then the file is not generated

> Does the rust test work?

yes, very much

mullroy, 2024-05-17:
The build system is quite daunting. The code in vendor/micropython seems to be both processed by GCC and Python ( & RUST too?). Therefore, overriding the warnings only in the CFLAGS isn't picked up by the python side. 
Doing the warning override directly in the source files fixed the problem:

vendor/micropython/mpy-cross/main.c
&nbsp;&nbsp;#pragma GCC diagnostic push
&nbsp;&nbsp;#pragma GCC diagnostic ignored "-Wenum-int-mismatch"
&nbsp;&nbsp;&nbsp;uint mp_import_stat(const char *path) {
&nbsp;&nbsp;&nbsp;#pragma GCC diagnostic pop
&nbsp;&nbsp;&nbsp;(void)path;
&nbsp;&nbsp;&nbsp;return MP_IMPORT_STAT_NO_EXIST;
}

vendor/micropython/py/stackctrl.c
void mp_stack_ctrl_init(void) {
&nbsp;&nbsp;&nbsp;volatile int stack_dummy;
&nbsp;&nbsp;&nbsp;#pragma GCC diagnostic push
&nbsp;&nbsp;&nbsp;#pragma GCC diagnostic ignored "-Wdangling-pointer"
&nbsp;&nbsp;&nbsp;MP_STATE_THREAD(stack_top) = (char *)&stack_dummy;
&nbsp;&nbsp;&nbsp;#pragma GCC diagnostic pop
}

Building the firmware
-----------------------------
cd core
make build_firmware
...
...
Fingerprint: c76ef17c4af347f0025be4e2daebf71b53899106ed7bf77990a3c774b36bb5dd
Signature is DEVEL, hashes are VALID