trezor/trezor-firmware #3854: rounded rect rendering bug

TychoVrahe, 2024-05-23:
When the rect height is smaller than 2*radius+2, the middle portion of the rect is blended three times, resulting in very different color.

possibly, width smaller than 2*radius+2 could break this too.

cepetr, 2024-06-07:
We have few options how to solve it:
1. Clamping: If the specified corner radius is too large, we will automatically reduced it so that it does not exceed the smaller dimension of the rectangle. 
2. Fallback to rectangle: If the radius is too large, we will draw a simple rectangle with no rounded corners.
3. Raise Error: This can help developers to catch this issue and fix it.

TychoVrahe, 2024-06-07:
I was thinking about 4) keeping the radius and rendering a render a sharp corner in the middle, but 1) seems better. 

We could also enable 3) just in debug, otherwise 1)