trezor/trezor-firmware #3858: Send ETH (EVM)

Hannsek, 2024-05-24:
**Design:** [Figma](https://www.figma.com/design/n43caWJFGJlT167AQcxruu/%E3%80%94-Prod-%E3%80%95Safe-5?m=auto&node-id=26%3A4182&t=OdRrLVz4xK6xA634-1)

ibz, 2024-08-14:
Here are screenshots from the other screens of the flow, as they are now in my PR...

[image]

Note: *Account info* screen missing for now, since we don't have the account # and wallet info...

[image]

[image]

[image]

[image]

[image]

[image]

obrusvit, 2024-08-14:
@ibz yes, use the current implementation of success screen. They can fit more text.
Figma success screens are a little bit outdated.

closed 2024-08-27 by ibz

ibz, 2024-09-03:
BTW, I reused this new flow also for the staking part, which now looks more like the Figma: https://www.figma.com/design/n43caWJFGJlT167AQcxruu/%E3%80%94-Prod-%E3%80%95Safe-5?node-id=160-368&node-type=CANVAS&t=q6Wjn67DNiTujwTm-0

Linked the relevant PR to this issue...