trezor/trezor-firmware #3859: Persistent words in recovery

Hannsek, 2024-05-24:
**Describe the bug**
When a word (e.g. 3rd) is deleted and user goes to the previous word (2nd), change the 2nd word and go to the next word (3rd) the word is present there.

**Firmware version and revision**
Core

**To Reproduce**
Steps to reproduce the behavior:
1. type 3rd word in recovery and confirm it
2. Delete the word and go to 2nd word, change it and confirm it
3. 3rd word is be present

**Expected behavior**
Once the word is deleted it shouldn't appear again.

Hannsek, 2024-08-01:
@trezor/qa can you please try if this is still happening? IMO it should be fixed by reworking the flow with TS5 but I'm not sure now and I cannot test it now by myself. :/

bosomt, 2024-08-01:
**QA NOK**

im sorry its still there :(
used 
academic again academic academic academic seed...
returned to `again`
after i filled in `again` `academic` is prefilled

https://github.com/user-attachments/assets/4476b568-bb86-4125-a995-6029195f1444

**Info:**
 - Suite version: web 24.8.0 (6016db166cafd3d2fd2cae67cafe95e9e08cd3f3)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:128.0) Gecko/20100101 Firefox/128.0
 - OS: MacIntel
 - Screen: 1512x982
 - Device: Trezor T3T1 2.8.1 regular (revision 55473a0b4be7b22317cd0c401c938d430d86f29f)
 - Transport: BridgeTransport 2.0.32

obrusvit, 2024-08-17:
@Hannsek I've looked into this but the requirement is unclear and it's a little bit more complicated:

## Currently
There are two ways how to go to previous word:
- delete the input field and press the "Previous" button (TS3, TS5)
- swipe from the left to the right (TT, TS5)

That means, TS5 has two options (swipe is inherited from TT). The swipe movement is interesting because it allow to go back multiple words without affecting them. So if you are at 6th word, you can swipe right to the 3rd word, rewrite it, and "confirm" again the pre-filled 4th and 5th word. Current behavior when going to previous word:
- TS3 - wrong, a user deletes the word but it is persistent
- TT - intuitive, a user does not delete anything
- TS5 - intuitive on swipe, wrong on "Previous" button

## Possible solution:
1. Keep the swipe behavior but distinguish whether the user go to previous word with something pre-filled or whether the input field is empty. Then use whatever was there previously as pre-fill.
    - TS3 - problem solved
    - TT/TS5 - problem solved, behavior of swiping would be preserved and deleting everything and going back would be intuitive
3. Always delete the `ith` word if you go to `(i-1)th` word 
    - TS3 - problem solved
    - TT/TS5 - swipe would delete everything
5. Always delete the `ith` word if you go to `(i-1)th` word and disable going back by a swipe movement on TT and TS5
    - TS3 - problem sovled
     - TT would have to get a new button to "go back" if the input field is empty, similar to TS5
     - TS5 - problem solved
     - TT/TS5 would lose the ability to go back multiple words and fix them

Hannsek, 2024-08-18:
The swipe on TS5 shouldn't be there as it might lead some users into unexpected situations. I'd choose number 3 but keep the swipe action on TT (which would delete the `ith` word)

PR #4110 (2024-08-28): Improve behavior when going to previous word during recovery
This PR improves behavior when going to previous word during recovery process. It

- runs the check for all keys in the keyboard if their press cannot lead to a valid word if the keyboard is initialized with a pre-filled word. Typically it means that all keys are disabled when going to previous word (always for SLIP39, most of the time for BIP39)
- deletes the word if the user goes back, related discussion: https://github.com/trezor/trezor-firmware/issues/3859#issuecomment-2294877348 
- remove the option to go back by swipe movement  [... cut]