trezor/trezor-firmware #3863: Show last typed PIN number for short period of time before changing it to "*"

Hannsek, 2024-05-26:
We have something similar in SLIP39 keyboard.

bieleluk, 2024-11-01:
@Hannsek I am not sure if this issue is relevant
The behavior depends on the layout model:

- `model_tr` does not have this issue. However, the time period of the last number being is bounded by the next button press (not a time period). 
[image]

- `model_tt` and `model_mercury` do not show the last digit but rather show an entire PIN when the area with pin dots is touched. However, there is nothing that informs the user about this feature. 
[image]
The only way to theoretically display this information is in the PIN dots are before the first digit is inserted.

obrusvit, 2024-11-01:
@bieleluk the goal here is to unify the behavior across models as much as possible. That means the following:
### `model_t` and `mercury`
1. after entering a digit, keep it revealed for a short period of time - e.g. a timeout of 1 second
2. reveal the whole PIN when the input area is touched (this is done)

### `model_r` 
  1. here, that's a question, I didn't know the digit is revealed until you move the carousel. I think a good modification would be to hide the digit after the "shuffle" of the carousel is finished. The resulting behavior would be similar to `model_t` and `mercury`. However, the implementation might be shared for this and the "Show" button so splitting it might be a little bit difficult.
  2. reveal the whole PIN when the "Show" button is pressed until you move the carousel (this is done)

### Summary
IMHO, the best effort is to newly implement the point 1. for `model_t` and `model_mercury` and keep `model_r` unchanged.

closed 2024-11-18 by bieleluk