trezor/trezor-firmware #3866: Check backup - wrong Suite response

Hannsek, 2024-05-28:
**Describe the bug**
When user press both buttons instead of the right button at the end of check backup flow, he'll get this error in Suite even though the check was successful.
[image]
[image]

**Firmware version and revision**
2T2B1 - 2.7.0

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Suite

**Expected behavior**
Success in Trezor Suite

ibz, 2024-07-23:
I am going through this flow, entering word after word... after the final word is entered, I get this screen:

[image]

Now I would need to press the "right" button, but it seems that if I press both, the flow is canceled. The device replies with:

[image]

So this might be the issue. I should probably make it so that the flow cannot be canceled anymore after all words have been entered.

Hannsek, 2024-07-24:
yeah, we should just restrict canceling in this place.