trezor/trezor-firmware #3880: walletCreation/Backup flow changes

Hannsek, 2024-06-02:
- [ ] [add copy to the third screen ](https://www.figma.com/design/n43caWJFGJlT167AQcxruu/%E3%80%94-Prod-%E3%80%95Safe-5?node-id=5706-7185&t=v4bQpNUZWu6zEBYO-4)
[image]

- [ ] [add text to the header of the word that is the same as previous one](https://www.figma.com/design/n43caWJFGJlT167AQcxruu/%E3%80%94-Prod-%E3%80%95Safe-5?node-id=5706-6990&t=v4bQpNUZWu6zEBYO-4). For example, 3rd and 4th words are the same (`academic`), display this header on the 4th word screen.
[image]

obrusvit, 2024-06-04:
1st point added in 5d68cc588e9ca4d4c3b8a8ec5a10d74d4ade21a6

obrusvit, 2024-06-24:
2nd point part of #3954

closed 2024-06-30 by matejcik